emsec/hal #46: Python functions cannot access imports and global objects

RenWal, 2019-05-02:
In this code snippet:

```python
import itertools
itertools
def test():
  itertools
test()
```

The call inside the function throws a NameError. The same holds for accessing global variables. That includes functions, meaning you can't access a function from within a function.

swallat, 2019-05-03:
Do you call it via the python shell or the editor?

RenWal, 2019-05-03:
This happens for me in both the shell and the editor:

```
NameError: name 'itertools' is not defined

At:
  <string>(5): test
  <string>(6): <module>
```

RenWal, 2019-05-06:
This is a workaround for the issue:

```
import itertools
itertools
def test():
  global itertools
  itertools
test()
```

Forcing imports (or functions or variables) to be global allows read access. However, in Python `global` should only be needed for write access to global objects, not for read access.

swallat, 2019-05-10:
@devhoffmann I think we should address this issue soon. I'll later take a look on this issue but might need some help.

swallat, 2019-05-10:
I think I have fixed the issue. We still need to test it before we release it again. @nils1603 could you maybe test the projects again with this new version?

closed 2019-05-10 by swallat