emsec/hal #65: Feature: In the navigation list, enable use of arrow keys

RenWal, 2019-05-10:
Currently, arrow keys do nothing in the navigation view. For easier navigation, allow use of the up and down arrow keys.

RenWal, 2019-05-13:
This seems to be a bug. See this snippet:
https://github.com/emsec/hal/blob/03c7df6d9d370e031842841ca9635f24c1380cb4/src/gui/graph_navigation_widget/graph_navigation_widget.cpp#L39
I'm not sure why this is there, but I assume it is because this is a `content_widget` that encapsulates a `QTreeView`. However, disabling focus also disables all key bindings, which seems to also disable the key bindings on the tree view. Removing that line fixes the issue, but we should ask the original author (git blame has no data on this) why they put that line there in the first place.

devhoffmann, 2019-05-14:
Agreed, this can be changed.
However the whole navigation list will be removed/replaced soon by a tree-like view

swallat, 2019-05-14:
@Slowpoke42 Could you respond to this?

Slowpoke42, 2019-05-14:
Yeah you can throw it out if you want, but as @devhoffmann mentioned the current navigation is largely obsolete and improvements made to it will likely be deleted with it.

Thank you for pointing this out anyway, keep it up :)

RenWal, 2019-05-14:
Had this locally anyway, here you go.

PR #76 (2019-05-15): Fix disabled arrow keys in navigation view
Remove the no-focus policy from the tree view to enable the key bindings. This allows navigation with all four arrow keys.
This PR closes #65.
closed 2019-05-15 by swallat