emsec/hal #71: HAL eats full CPU on one core to paint some QT elements

RenWal, 2019-05-13:
After some time, HAL starts to consume 100% CPU on one core. Attaching a debugger yielded this call stack at the moment of stopping the offending thread.

```
libhal_gui.so!gui_graph_gate::paint(gui_graph_gate * const this, QPainter * painter, const QStyleOptionGraphicsItem * option, QWidget * widget) (/home/rene/git/hal/src/gui/graph_layouter/gui_graph_gate.cpp:212)
libQt5Widgets.so.5![Unknown/Just-In-Time compiled code] (Unbekannte Quelle:0)
libQt5Widgets.so.5!QGraphicsView::paintEvent(QPaintEvent*) (Unbekannte Quelle:0)
libhal_gui.so!graph_layouter_view::paintEvent(graph_layouter_view * const this, QPaintEvent * event) (/home/rene/git/hal/src/gui/graph_layouter/graph_layouter_view.cpp:292)
libQt5Widgets.so.5!QWidget::event(QEvent*) (Unbekannte Quelle:0)
libQt5Widgets.so.5!QFrame::event(QEvent*) (Unbekannte Quelle:0)
libQt5Widgets.so.5!QGraphicsView::viewportEvent(QEvent*) (Unbekannte Quelle:0)
libQt5Core.so.5!QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) (Unbekannte Quelle:0)
libQt5Widgets.so.5!QApplicationPrivate::notify_helper(QObject*, QEvent*) (Unbekannte Quelle:0)
libQt5Widgets.so.5!QApplication::notify(QObject*, QEvent*) (Unbekannte Quelle:0)
libQt5Core.so.5!QCoreApplication::notifyInternal2(QObject*, QEvent*) (Unbekannte Quelle:0)
libQt5Widgets.so.5!QWidgetPrivate::sendPaintEvent(QRegion const&) (Unbekannte Quelle:0)
libQt5Widgets.so.5!QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (Unbekannte Quelle:0)
libQt5Widgets.so.5!QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (Unbekannte Quelle:0)
libQt5Widgets.so.5!QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (Unbekannte Quelle:0)
libQt5Widgets.so.5!QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (Unbekannte Quelle:0)
libQt5Widgets.so.5!QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (Unbekannte Quelle:0)
libQt5Widgets.so.5!QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (Unbekannte Quelle:0)
libQt5Widgets.so.5!QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (Unbekannte Quelle:0)
libQt5Widgets.so.5!QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (Unbekannte Quelle:0)
```

swallat, 2019-05-14:
@RenWal Could you elaborate a bit more on when this issue occurs?

RenWal, 2019-05-14:
So far this looks rather random to me (which is not really helpful). I'll report back if I can pinpoint an exact action that triggers this.

Slowpoke42, 2019-05-14:
@RenWal Don't bother. If the problem is related to the current graph view it's likely not worth your time, because I'm currently rewriting that feature from scratch. The graph view classes used in the master branch (layouters, items, scene, view and widget) are essentially obsolete and serve more of a placeholder function. In the unlikely event that this problem is still present in the new version we can reopen this issue.

swallat, 2019-05-15:
I'm closing this issue for now.

closed 2019-05-15 by swallat