emsec/hal #78: Feature: Expose gate init string in Python

RenWal, 2019-05-16:
For many tasks one needs to extract the truth table of a gate. Currently, the BDD must be extracted via a gate decorator and then evaluated to build the truth table.
However, the entire truth table is already present in the VHDL code in form of the init string of all LUT gates. I suggest exposing the init string as a property of each gate to simplify the process of truth table extraction.

RenWal, 2019-05-22:
This also applies to automatic FSM extraction using init strings of Flipflops.

swallat, 2019-05-23:
I actually think we could do that. @devhoffmann What do you think?

swallat, 2019-05-23:
We could think about including it in the gate details widget, though.

RenWal, 2019-05-23:
Yes, this is what I meant. Is it possible that this feature is currently undocumented?

devhoffmann, 2019-05-23:
The feature is not really undocumented...
The problem is that for example ASIC netlists do not contain INIT values. 
Hence we needed a mechanism to store additional information for arbitrary architectures.
We decided to have all our netlist elements inherit from a data container class which offers this `data` field in python.
In the end, all additional (i.e., non generic) information is stored in there.
As this also applies to init values, but those are not generically available for all kinds of netlists there is no specific documentation for these values, but for accessing the additional data in general

swallat, 2019-05-23:
Should we add an example in the python documentation?

swallat, 2019-05-23:
Hm `get_data` is missing in the python documentation.

PR #99 (2019-05-29): Fixed correct python binding of data container and added documentatio…
…n for it. Closes #78
commit 5e6fcd65 (2019-05-29): Fixed correct python binding of data container and added documentation for it. Closes #78
commit 65ec318b (2019-12-05): Merge branch 'master' into test/test_update

* master: (29 commits)
  Update README.md contact from gitter to slack
  Bugfix/python editor searchbar behavior (#116)
  Move from Discord to Slack
  Limit Block count in python console QTextEdit
  Updated Changelog v1.0.20
  Add overwrite function for python editor
  Enable closing prompt for unsaved python tabs
  Feature/python editor tabs (#107)
  Update Changelog
  Remove dirty flag from genversion.py
  Remove kernel.h dependency in include/gate_decorator_system/decorators/gate_decorator_bdd.h for plugin repositories
  Update Changelog
  Fix un [... cut]
closed 2019-05-29 by swallat