emsec/hal #85: Naming of insertion methods in Python class "module" inconsistent

RenWal, 2019-05-22:
The Python class "module" contains the methods `insert_gate` and `add_net`. This is unintuitive, so I suggest renaming one of the two methods.

Slowpoke42, 2019-05-22:
Since the latest rework module membership is no longer optional. Every gate / net is part of one and only one module at all times. Inserting a gate / net into a module is consequently synonymous with removing it somewhere else. The current interface does not communicate this behavior to the user. Maybe a different verb like "move" or "relocate" would be more fitting.

swallat, 2019-05-23:
@Slowpoke42 I think `move_gate` and `move_net` would be good. @devhoffmann What do you think?

devhoffmann, 2019-05-23:
I suggest going for `assign`

swallat, 2019-05-23:
Hm, `assign` is better. It makes sure that we add the net or gate to the current module. `move` alone does not make clear where we move the gate or net.

PR #95 (2019-05-29): Rename add_net and insert_gate to assign_net and assign_gate in modul…
…e class. Closes #85
commit d470f878 (2019-05-29): Rename add_net and insert_gate to assign_net and assign_gate in module class. Closes #85
commit 65ec318b (2019-12-05): Merge branch 'master' into test/test_update

* master: (29 commits)
  Update README.md contact from gitter to slack
  Bugfix/python editor searchbar behavior (#116)
  Move from Discord to Slack
  Limit Block count in python console QTextEdit
  Updated Changelog v1.0.20
  Add overwrite function for python editor
  Enable closing prompt for unsaved python tabs
  Feature/python editor tabs (#107)
  Update Changelog
  Remove dirty flag from genversion.py
  Remove kernel.h dependency in include/gate_decorator_system/decorators/gate_decorator_bdd.h for plugin repositories
  Update Changelog
  Fix un [... cut]
closed 2019-05-29 by swallat