emsec/hal #86: Feature: Add option "show only this module" in graph view

RenWal, 2019-05-22:
Suggestion for the dev working on the new graph view (if this is not already planned):

When working with large netlists where a functional unit (e.g. an FSM) is spread out across the entire graph, it is often hard to figure out how the components interact. I suggest adding a toggle "show only the selected module" that takes all gates and nets in the module (selected in the navigation list) and layouts them in a new, small graph.

This would allow for much easier visual inspection of the connections and you could increase the zoom level without something going off-screen.

Slowpoke42, 2019-05-22:
Yup, already working on it. You can check out branch feature/new-graph-views if you want to take a look at it, but its obviously WIP code, meaning most of the components are not finished / connected, so dont expect to see much.

swallat, 2019-11-04:
@Slowpoke42 Is this still Work in Progress?

swallat, 2019-11-04:
Closed in favor of #195

closed 2019-11-04 by swallat