emsec/hal #87: Feature: Working with net parts

RenWal, 2019-05-22:
Currently only whole nets can be selected and added to modules. This can cause cluttered module views.

Minimal example: In VHDL, create a small FSM from a FlipFlop and an inverter and then add 20 connections from the FF output to other gates. Now, try to add the nets connecting the FF and Inverter to a module. You will end up with a cluttered graph view with all the connections to other gates because they form one single net (because they are all fed from the same FF output pin).

I suggest adding an option to either splice nets or to add net parts to a module (in the form of "add this net to the module, but only the part to this destination"). This would de-clutter the graph view a lot, giving the reverse engineer the ability to focus better on the components of interest.

swallat, 2019-05-23:
We are currently working on a Cone view, to support that. It should also be included in the branch feature/new-graph-views.

swallat, 2019-11-04:
Closed in favor of #195

closed 2019-11-04 by swallat