emsec/hal #90: List of issues found while testing

FabianAlbertRub, 2019-05-24:
List of issues found while testing:
- netlist/netlist::is_module_in_netlist: only checks the id not the pointer. Therefore we can pass modules with the same id, that aren't part of the netlist anymore to get a false  or pass a nullptr, which results in a segmentation fault.

- netlist/module::set_name: does not check whether the passed name is the empty string. At the other hand, the create_module function (of the netlist) does not allow the empty string.

- netlist/netlist_internal_manager::create_module: The module created by the create_module, may have a parrent owned by another netlist

Slowpoke42, 2019-05-24:
Hey, thanks for the report. Two minor problems:
First, unless your set of issues is inseparably linked and should only be discussed / solved together please open a new github issue for every single one of them. I know it's tedious but it's just much easier for us to organize and keep track of them.
Second, please give your issues descriptive names. That is obviously easier if an issue only contains a single question/request/bug so this is somewhat related to the first point. "netlist::is_module_in_netlist doesn't check the pointer" for example would be a descriptive title.

This might seem pedantic, but issues usually appear quicker than they are solved so at some point we might have 500 of them simultaneously and if they're all called "list of issues", and contained 3 unrelated problems, this tracker would be next to useless.

I'll probably add some kind of instructions on how to open an issue in the near future.

swallat, 2019-05-29:
I agree with @Slowpoke42. Could you split it in individual issues?

swallat, 2019-05-29:
Closed in favour of #96 #97 #98

closed 2019-05-29 by swallat