emsec/hal #102: Python input() expects data on wrong console

RenWal, 2019-05-30:
**Describe the bug**
Calling `input()` in either the Python console or editor locks up the GUI (see #101) and one can't type their input into the console as they would expect from the behavior of the standard python interactive shell. However, if HAL is started from a command line, typing something in that terminal and pressing [Enter] will pipe your input into the Python script.

**To Reproduce**
1. Start HAL from the command line with `hal -g`
2. Type `input()` in the Python console or run it from the editor
3. See that the GUI becomes unresponsive and you can't type into the console
4. Pull up the terminal you started HAL from, type something and press [Enter]
5. See that the GUI is working again.
6. If you typed into the Python console and not the editor, you can also see immediately that `input()` received whatever you typed on the command line.

**Expected behavior**
A user will probably have some experience with the Python interactive shell and expect the HAL Python console to work the same way. Calling `input()` should allow the user to type their input into the Python console (and not the system's terminal).

swallat, 2019-06-03:
If this is an issue, we could override the python function to allow this behavior. Could you give us an example where you used it?

RenWal, 2019-06-03:
```
gateName = input("gate to print > ")
gates = netlist.get_gates(name_filter=gateName)
for g in gates:
    print(g.get_name() + ' has these inputs:')    
    for net in g.get_fan_in_nets():
        print(' ' + net.name)
```

Of course this is constructed and doesn't really have much practical use, but it demonstrates what I want do be able to do. Especially if you have larger scripts, some interactivity is nice, e.g. to select which analysis steps to execute.

swallat, 2019-06-06:
Ah ok. We might be able to do that by reimplementing the input function within our library. We do basically the same to catch the python output. Though, I might not be sure, if we are able to that without blocking the complete GUI.

joern274, 2022-10-17:
Fix merged into master

closed 2022-10-17 by joern274