emsec/hal #105: VHDL writer writes single bits wrong

nils1603, 2019-05-31:
FDRE_0 : FDRE
  generic map(
   INIT => X"0"
  )
 port map (
   C => signal_0,
   CE => signal_104,
   D => signal_12,
   R => signal_103,
   Q => signal_99
);

has to be... INIT => '0'

might be an error in the parser?

nils1603, 2019-05-31:
Error is in Verilog parser. is not read as a bit value

swallat, 2019-06-13:
@nils1603 Do you have a specific netlist to use for testing?

nils1603, 2019-06-13:
error occurs in any Verilog library when a single bit is read to a bit vector

swallat, 2019-06-19:
Hm, we are probably not able to fix this in HAL. Verilog does not provide a data type for a single bit. So the parser actually works correctly. We might be able to do postprocessing in the writer, but we would need to do that based on the gate type. So we cannot fix it for now.

closed 2019-06-19 by swallat