emsec/hal #106: Netlist parsing failures don't trigger an error dialog

RenWal, 2019-06-02:
**Describe the bug**
If deserialization fails with all gate libraries when opening a file from the welcome screen, no error message is shown and the GUI simply stays on the welcome screen.

**To Reproduce**
1. Start HAL from terminal
2. Open a file that is not parseable with any installed gate library
3. See that nothing happens
4. Look in the terminal and see that parsing failed with all libraries

**Expected behavior**
An error message should be displayed to the user.

swallat, 2019-06-03:
I agree we should make error messages more visible.

dertob, 2019-06-06:
If I see correctly there are only a few errors that can occur before a file has been loaded and therefore can not be displayed in the GUI's logger. **I suggest to simply display a message box containing the error message if it's one of these few cases.** All the other errors messages after successfully opening a file should be visible in the GUI's logger with no problem.

In the future we can discuss about all errors triggering a message box. That means the event system must be extended with error events though. In my opinion this is not necessery yet. Displaying the few errors mentioned above should be enough for letting the user know what's going on.

What's your take on that @swallat?

swallat, 2019-06-11:
For a quick fix, I would agree with your solution. Anyway, we should think about a better error notification system soon. @dertob Do you take care of the quick fix?

dertob, 2019-06-11:
@swallat Yeah, I'll take care of the quick fix.

swallat, 2019-06-13:
@dertob Do you think you finish this today, or more likely later. In other words, do you complete it in time for today's release?

dertob, 2019-06-13:
Unfortunately I have no time at all to deal with the issue this week. @swallat

swallat, 2019-06-13:
@dertob No problem. Thanks for the info.

PR #133 (2019-06-19): Add message box for error while opening file
Fixes #106
closed 2019-06-19 by swallat