emsec/hal #124: Empty VHDL Port Map crashes HAL

swallat, 2019-06-18:
**Describe the bug**
When parsing an incorrect VHDL netlist, with empty port maps of gates HAL crashes.

**Expected behavior**
hal should print an error and continue working.

swallat, 2019-06-18:
@devhoffmann What would you suggest? Error message or crash?

devhoffmann, 2019-06-18:
I think we should depend on correct input, i.e. on any error we abort parsing and return a nullptr which will result in an error message.

RenWal, 2019-08-07:
I think this is happening in this piece of code:
https://github.com/emsec/hal/blob/777a911ad2a0d11dc4c5b983812d6e5ea6a5a478/src/hdl_parser/hdl_parser_vhdl.cpp#L437
The code expects either a generic map or port map and sets `pos` accordingly. If neither map exists, then `type.substr(pos)` with pos=MAX_INT is called, causing substr to abort.

Minimal example for the crash:
```
begin
end;
```

swallat, 2019-11-04:
@RenWal @devhoffmann What is the state of this issue?

RenWal, 2020-02-14:
I can't reproduce this issue anymore and assume it has been fixed as a side-effect of restructuring the VHDL parser. @swallat do you happen to still have the netlist that caused the crash back when you reported this? If so, do you mind re-testing against the current master?

closed 2020-10-15 by SJulianS