emsec/hal #136: recent-file-item stays after deleting / moving a file and then try to open it

ChuTriel, 2019-06-23:
**Describe the bug**
After moving/deleting a .vhd or .hal file that is listed in the recent files widget and you try to open it via the recents file widgets again, hal opens a dialog that says it could not open the file, yet the item remains in the list. It is also currently not possible to remove an item from the list. It is possible that the list gets infinitely big.

**To Reproduce**
1. Open a .vhd or.hal file
2. Move the file to another folder
3. Try to open the file via the recent files widgets

**Expected behavior**
Either ask wether the entry should be removed or stays, or for the simplicity, just remove the entry from the list.

RenWal, 2019-06-24:
I suggest we simply check whether the file exists on startup and flush any entries from the list that do not exist on disk. If we further limit the recent files list to, let's say, 20 entries, this should not incur a significant slowdown of the startup process.

ChuTriel, 2019-06-24:
Thats certainly one way of doing it, yet when you look at other project-based applications such as Visual Studio, PyCharm etc the entrys stay and you will be notifed that it is missing when you try to open the project. Your suggestion should not result in a noticeable delay when you start the gui. I will look further into this.

RenWal, 2019-06-24:
To be honest, I find the way VS deals with defunct recents utterly annoying. I don't want to be offered "here, I remembered this file for you", click it and then be told "whoops, it's gone". If we really, really want to keep the list intact (i.e. not automatically purge entries) then I'd suggest disabling these entries (make them unclickable and change their color) so the user experience is fluent.

swallat, 2019-06-24:
I think it is also better when it stays in the list but is marked as missing. Consider following situation:

1. You delete your cloned repo with all netlist and Hal files in it. It might have. Even corrupted and you need to reclone the folder.

2. You Start HAL for another netlist.

3. After that you remember to clone the repo and continue your work. Here it would be nice to directly open your netlist from the recently opened menu.

What do you think?

ChuTriel, 2019-06-24:
Alright, sounds plausible.

closed 2019-07-04 by ChuTriel