emsec/hal #152: Python allows setting invalid init strings

RenWal, 2019-06-27:
**Describe the bug**
On a LUT6 you should set an init string this way:
```
lut.set_data("generic", "INIT", "bit_vector", "0"*16)
```
However, doing this also works:
```
lut.set_data("generic", "INIT", "bit_vector", "0"*4)
```

This causes all sorts of issues, including plugins that rely on correct config strings not working.

**To Reproduce**
Steps to reproduce the behavior:
1. Take any netlist with a LUT6.
2. Run the second code snippet.
3. See that HAL does not prevent this.

**Expected behavior**
HAL should not let the user do this if the init string clearly does not match the gate type.

**Additional context**
This works for other gates as well.

devhoffmann, 2019-06-28:
I don't see this as a bug.
HAL does not have any information on the kind of gate and the data map can be used to store arbitrary data.
During parsing, all "unknown"/unused information simply gets put into the data map and for LUTs/FFs this happens to be the init string.
If the user makes a mistake at this part, it is his fault...

swallat, 2019-06-28:
Do we have a way to set new data via the decorator? If not, we should add a function there. Furthermore, I would agree with @devhoffmann. `set_data` is a function to append arbitrary data to a gate.

nils1603, 2019-07-01:
Agreed. Not a bug, but a feature that - depending on the library - might have use.

devhoffmann, 2019-11-04:
Closed as the discussion came to a conclusion.

closed 2019-11-04 by devhoffmann