emsec/hal #156: VHDL parser should be able to parse a netlist that is not flattened

devhoffmann, 2019-06-28:
**Is your feature request related to a problem? Please describe.**
With the availability of modules, the vhdl parser should be able to parse a netlist with hierarchy.

**The solution we will implement**
All entities are parsed separately before inserted into the top entity as modules.
Signal names on inputs/outputs can be taken from the parent entity, internal signals are prefixed with the entity name.

swallat, 2019-06-28:
I strongly support this feature.

devhoffmann, 2019-07-05:
Update: The feature is basically done in its respective feature branch.
It now needs testing.

swallat, 2019-07-05:
@devhoffmann: Any suggestions how we should proceed?

devhoffmann, 2019-08-15:
Now merged into the master and used as a default.
The old parser is still available using the ´--language vhdl_old´ switch

closed 2019-08-15 by swallat