emsec/hal #157: Global content manager tracking is faulty

dertob, 2019-06-30:
**Describe the bug**
The global content manager doesn't track python tabs correctly. Depending on the position of the tab in the bar the script gets wrongly identified as saved or unsaved. This leads to wrong notifications popping up and eventual loss of data.

**To Reproduce**
Steps to reproduce the behavior:
1. Open 2 python script tabs in the editor e.g. 'a.py' and 'b.py'
2. Fill both tabs with some code and save both
3. Modify the content of 'a.py'
4. Move the tab 'a.py' to the right, thus switching the position with 'b.py' in the tab bar
5. Click on tab 'b.py' and save tab 'b.py'
6. Close HAL
7. HAL closes without warning, despite changes occurring in 'a.py'

**Expected behavior**
HAL should warn the user that the content of a.py is unmodified.

 It should for now at least. For the future the proposed solution from #142 should be considered.

**Additional context**
The way the changes are tracked in the content manager seem to be dependent on the index of the tab. This seems to be the main issue on first look.

devhoffmann, 2019-07-08:
@dertob What is your status on this? Can you explain your solution strategy?

dertob, 2019-07-08:
I think that the existing idea of a global class for tracking changes on files across all widgets is good and more elegant than trying to pass signals through multiple layers. My plan is to implement a new class with the same functionality provided in the current 'content manager'. To fix the issues we have with the current tracking I would equip each instance of the 'python_code_editor' with a uuid provided by Qt and safe that instead of just the file name or the tab number. Also one could argue to put the functionality from 'netlist_watcher' also in the new class. It's whole purpose is to track changes on the netlist.

My current plan is to implement this tomorrow.

PR #167 (2019-07-12): Fix/global-content-manager
Fixes #157.
closed 2019-07-12 by swallat