emsec/hal #161: bugs and unnecessary code parts in hdl_writer/hdl_writer_vhdl.cpp

FabianAlbertRub, 2019-07-05:
- the if-block in l. 67:
`if (std::all_of(net_temp.begin(), net_temp.end(), ::isdigit)) {...}` 
can't be entered, since the `get_net_name` function already adds the "NET_" prefix to a net with only digits.

- consider the following lines of code (l. 372 ff.):
   ```
   if (std::get<1>(d.first) == "loc" || type == "time")
       continue;
   if (type == "time")
   {...}
   ```
   The second if block can not be entered, so generic time data will not be handled correctly

- the function `get_port_name(std::string pin)` is unused, so it can be removed. Or will it be used in later development?

(Since these are only small issues, which can easily fixed by only one person, i wrote them together in one report ;))

nils1603, 2019-07-10:
This reallyyy needs work! We need to write tests. Is there any VHDL parser that could be used to parse our written VHDL again? We could use the parser to write tests!

RenWal, 2019-07-10:
Can we implement basic self-testing by looping back our VHDL output to our own VHDL parser?

FabianAlbertRub, 2019-07-11:
Yeah, I've already written tests for the writer, based on comparisons with our parsers input(this thread is actually based on my testing observations).
To check the output for every call of the writer may be an interesting idea, but can result in worse run-time (especially for bigger netlists). At the other hand it seems to be a nice idea to feed the tests with more data. Maybe we can discuss this in our next meeting ;)

nils1603, 2019-07-11:
We could do that, but our parser is also excepting VHDL code that is 100% accurate regarding the specification.

closed 2021-06-15 by SJulianS