emsec/hal #164: Add close-all button in the Python editor

RenWal, 2019-07-08:
**Is your feature request related to a problem? Please describe.**
When you experiment with a lot of different scripts and then want to clean up, you need to close every tab manually.

**Describe the solution you'd like**
I suggest adding a close-all button to quickly clean up your workspace.

**Describe alternatives you've considered**
If you don't want to clutter the UI, I would suggest adding a context menu to the tabs (maybe like in VSCode with "Close all right", "Close all left", "Close all others", "Close all saved")

RenWal, 2019-07-08:
If we can agree on whether to use a context menu or a simple button (and with which options) then I can pick this issue.

steffenbecker, 2019-07-25:
I like the idea of the context menu since it seems to be the more intuitive approach compared to other UIs.

RenWal, 2019-09-29:
Closed in 16bf5f9

closed 2019-09-29 by RenWal