emsec/hal #166: Discussion: New Navigation Widget

Slowpoke42, 2019-07-08:
Because the current navigation widget wasn't really designed for, and as of now doesn't support, hierarchical modules, I am currently in the process of rewriting it from scratch.

As far as I can see there are three distinct approaches for this rewrite.

1. The entire netlist is displayed as a single big tree, with the topmodule as the root and the physical gates (and nets ?) as the leaves. This is conceptually similar to the way in which most IDEs display folder structures, see [https://qt.developpez.com/doc/4.6/images/qtcreator-sidebar.png](url)

Because the entire hierarchy is visible this approach provides structural overview and allows for very direct interaction. One could, for example, select 17 different gates / modules from all over the netlist and then drag & drop them into a new module with very little effort / overhead.

However, due to the huge size and complexity of most netlists this approach might not be optimal. If the entire tree structure is expanded, the user has to scroll and look through a structure of potentially tens of thousands of items for every little operation. This can be solved by allowing the modules to be expanded / collapsed but it requires a lot of micromanagement from the user to manually open and close items by clicking on the tiny arrow thing.

Even if the tree structure is properly managed it includes a lot of clutter. There is a high likelyhood that the irrelevant items would outnumber the relevant items for pretty much every task.
If, for example, you want to work in a module 5 levels deep, you by definition have to expand at least 5 probably unrelated modules, which are now cluttering up your navigation widget.

Lastly, because child items are indented, a tree structure grows horizontally with its depth. Depending on the average depth of the module hierarchy this could mean that the navigation widget would take up half the screen or more.

2. Only single modules can be displayed. The exact form in which they would be displayed is up for discussion, but it would likely include a list of all direct submodules, all directly contained gates (as opposed to indirectly contained in submodules), and potentially some or all related nets (?).

The module hierarchy could then be traversed by either opening a submodule, most likely by double clicking it, or by pressing the "module-up" key (backspace ?) which would take you back to the parent module.

Under this approach the navigation widget would work exactly like the new graph view, except for the fact of course, that the graph view displays the data as a graph while the navigation displays it as text (and probably some form of icons).

This is conceptually similar to how most file explorers work (when you ignore the tree that they usually include somewhere on the side). When you, for example, open a directory in windows explorer it only shows you the contents of that directory. You can then travel down the hierarchy by double clicking a folder, or back up to the parent folder with the back shortcut (alt + arrow up).

see [https://i1.wp.com/www.winhelponline.com/blog/wp-content/uploads/2019/05/file-explorer-not-highlight-files.png?fit=1038%2C671&quality=100&ssl=1](url), if you have never seen windows explorer.

While the overall structure of the module hierarchy is less obvious, working with individual modules becomes much easier, due to the fact that there is zero clutter and no indentation.

3. A hybrid solution. The navigation widget only displays modules hierarchically as a tree. A module item can then be activated (most likely double clicked ?) to open a list of its contents. The exact form and function of such a list is up for discussion, but it would likely contain all direct submodules, all directly contained gates, and potentially some or all of the related nets (similar to 2.)

This solution combines the structural overview of 1. and the relative simplicity of 2. but it's probably not the most natural solution to use...

I don't have a picture for this, so i'll leave it up to your imagination.

If you can think of some completely different approach that I overlooked here please let me know.

Of course it's not strictly necessary to pick a single solution, we could theoretically just implement all three of them and leave the decision up to the user, but that would still leave us with the decision of which to implement first.

If you have an opinion, please describe which of the solutions you prefer, why you prefer it, and how it should work specifically (notice that the descriptions I provided are fairly general and leave open questions).

If you want to show of your MS Paint skills, mockups would also be greatly appreciated.

RenWal, 2019-07-08:
I like the hybrid approach. If you have ever worked with the Eclipse IDE, it does a similar thing with its project explorer: The project structure is shown as a tree and you can expand infinitely. If this causes too many nested levels or you want to concentrate on a single folder, you double-click it. The folder then becomes the new top-level element and all parents are hidden.

This would provide the user with a good overview of all modules if they work on a simple project, but still allows them to de-clutter the interface with a simple double-click when they are working on something more complex.

Regarding nets and gates, what if we give the user a setting like "show gates in another widget"? The navigation widget would then split into a "modules" and "gates of selected module" thingy. That means the gates are visible if needed but they don't take up lots of space in the tree, pushing other modules out of view.

Slowpoke42, 2019-07-08:
Yea, I guess using a pure tree structure, but giving the user the ability to redefine the root element (or to hide all other elements) would be another option.

The question is whether or not its the right solution for the use case. When writing code you usually work fairly localized (same file or multiple files in the same directory) for prolonged periods of time. While writing the graph view, for example, I usually stay inside the same folder for 8 hours straight, so having to do one extra action to make that folder my main folder wouldn't be much of a bother, but I assume (I don't know) that the module hierarchy will be used less localized, at least when starting to work on a new netlist thats not neatly divided into modules yet. Could be wrong though.

Splitting the widget into multiple sections, one, for example, for the module hierarchy and another for the contents of the currently selected module is also possible. Its basically approach 3. except now the module contents widget doesn't replace the hierarchy view when a module is selected, but is instead placed next to it.

In some sense we already have that exact solution. I mean its not really implemented yet, but the details widget should technically also reacts to module selections and would, in that case, act more or less in the exact way you describe. I dont know what information the module details widget should display, but a list of its components seems like a good start.

We probably have to talk about the details widget separately at some point.

nils1603, 2019-07-19:
The hybrid solution should be the way to go. All gates not in a submodule should be in their own list in the submodule called "Uncategorized Gates"

- TopModule
    -  Submodule 1
          - Subsubmodule 1
    - Uncategorized Gates

All of them should be lists

RenWal, 2019-07-20:
I strongly agree on the Uncategorized Gates list and that we should avoid mixing lists and list items on the same level. In the current navigation, this confused me several times (albeit with nets, not gates):

[image]

Here, if you do not look closely, "N2" seems to belong to the Global Outputs. However, it is actually on the same level and only lacks the expand button because it is not a list but an item. The indentation makes this hard to distinguish.

closed 2021-06-15 by SJulianS