emsec/hal #171: Python files are marked as saved when the netlist is saved

RenWal, 2019-07-25:
**Describe the bug**
When saving a netlist as .hal file, all open python editors with pending changes are marked saved. This is only partially correct: When the HAL file is saved, the unsaved Python files are included, so no work is lost. However, the actual Python files are not saved individually.

**To Reproduce**
Steps to reproduce the behavior:
1. Open any netlist
2. Open a .py file in the python editor and type something, this will cause an asterisk to appear in the tab
3. Hit [Ctrl-S]
4. See that the asterisk is gone

**Expected behavior**
Users will expect the asterisk to disappear iff the actual script file has been saved.

**Additional context**
We should clearly document what gets saved where and when.

dertob, 2019-07-25:
I think we need to discuss when an asterisk has to be added to the tab or define what the asterisk actually means. Usally it means there are unsaved changes. Save the .hal and it's technically saved. In my opinion the whole system how py script are saved and associated with the .hal file is the cause of confusion here.

Slowpoke42, 2019-07-25:
I think python scripts should only be written to their individual source files. I do not know of any other editor that saves opened files twice, once in the actual file and once in some kind of project file (excluding temporary backups). Seems unnatural and overcomplicated to me.

The only benefit of the current system seems to be that the python scripts are still available if you move or delete (intentionally or accidentally) the source files, but...

a) How often does that happen? It seems kind of contrived.
b) Even if it does happen, it's not our problem. Thats the users problem. If you delete your files, your stuff is gone - thats how it works. If the user wants additional security he is free to use any backup or redundancy system he wants.

One could argue that there is the additional benefit of being able to share your entire project with other people, python scripts included, by sending them the project file, but once again thats a non problem. Zipping a folder takes like 3 seconds, we dont need a custom solution for that.

RenWal, 2019-07-25:
I have previously suggested splitting this saving mechanism between never saved files and modified files. The first ones go into the HAL file. The second only go to their script file, but HAL remembers their path in the HAL file. Thus, unsaved files are not lost and external modifications of scripts while HAL is closed don't collide with what is saved in the HAL file. If the user deletes the script, it's gone.

closed 2022-08-26 by SJulianS