emsec/hal #186: Graph view does not update when moving one or more gates to an existing module

RenWal, 2019-09-29:
**Describe the bug**
On branch integrate-new-graph-features, when right-clicking on a gate or a selection of gates, the user can move the gate(s) to a new or existing module. The latter does not update the view.

**To Reproduce**
Steps to reproduce the behavior:
1. open a netlist
2. create a module under the top module
2. in the graph view, enter the top module
3. right-click on a gate or a selection of gates
4. select "Move to module …" -> your created module
5. see that the change is not reflected in the graph view

**Expected behavior**
The graph view should update the way it does when you select "Move to module …" -> "New module …"

swallat, 2019-09-30:
@RenWal How do you create the module? Via the python interface?

RenWal, 2019-09-30:
You can do it that way. In my previous observation it does not make any difference whether you create the module via Python or the GUI.

```python
netlist.create_module("foo", netlist.get_top_module())
```

RenWal, 2019-09-30:
Confirmed fixed in integrate-new-graph-features.

commit efe32ab8 (2019-09-30): fixes #186
closed 2019-09-30 by RenWal