emsec/hal #189: Discussion: Things we need global settings for

RenWal, 2019-10-06:
The settings page and internal management are now ready for integration into the UI. To complete integration, I need to know which components need what kind of settings. **GUI developers, please comment below if you need to add your settings to the settings page (name, type, default value, value limits if applicable).**

Currently, these settings are planned:

- GUI style (unchanged from the old settings page)
- Python editor font size
- Python line number toggle
- Python minimap default state
- Graph grid on/off and grid type selection

Settings are represented as follows:

[image]

Currently, the following types of settings are available:

- checkboxes
- dropdowns
- text fields
- sliders

Please let me know if you require any additional types of settings.

Slowpoke42, 2019-10-07:
I have already marked a few variables as stuff that should be configurable through the settings with comments. Apart from that I don't really have a comprehensive list but here are some ideas:

- Default directories for everything. Netlists, plugins, HAL files, log files, stylesheets, and so on
- Auto save and auto save time interval
- Keybinds should be fully configurable at some point.
- Default layouter, gate, module, and net classes for the graph widget.

RenWal, 2019-11-04:
Implemented on _integrate-new-graph-features_ as of today:

- auto-save and interval
- keybinds for toolbars and menus (there are still some hard-coded keybinds that don't use QAction, I'll have to hunt them in the UI code at some point)
- Python editor options such as line wrap, highlight of current line, minimap default state, etc.
_____

If you want to introduce new keybinds on QActions, please do not hard-code them but use:
```CPP
g_keybind_manager.bind(m_action_foobar, "keybinds/mywidget_foobar");
```

This will automatically assign whatever keybind is configured at "mywidget_foobar" and keep the QAction updated when the user edits their configuration.

closed 2021-12-20 by SJulianS