emsec/hal #211: Modules don't show number of gates properly

nils1603, 2019-11-21:
Discussion needed:

I personally think that when you click on a module you should see the total number of gates in the module. If you click on a module right now and the module has "sub"modules it will show 0. We should fix that...

And the modules are not sorted like expected:
see screenshot
[image]

RenWal, 2019-11-26:
Now tracking the sort order problem in issue #214.
Splitting different problems into different issues helps us track progress and assignees.

devhoffmann, 2019-12-10:
This issue is solved in HAL 2.0

closed 2019-12-10 by devhoffmann