emsec/hal #231: Is the name HAL safe to use?

markusdd, 2020-01-04:
Hi, professional hardware designer here.

I followed the talk at CCC with great interest, but once I heard the name of the tool I immidiately
thought about work. Cadence Design Systems uses HAL as the Trademark name for their HDL Analyzer Tool as part of Incisive and Xcelium RTL/netlist simulator suites. It is a Linter which is used to uncover structural problems in the design like clock domain crossing, combinational logic in reset paths etc. and to enforce certain coding rules, including FSM coding styles etc.

As this is in a very similar domain I was wondering if it is safe for you to use this project name without hurting this trademark?

I'm not affiliated to cadence in any way, this jsut popped into my mind and I wanted to bring this to your attention. Might be totally irrelevant but I felt notifying you cannot hurt.

devhoffmann, 2020-01-04:
Thank you for pointing out! We will take a look at this with our team

swallat, 2020-01-06:
Thank you for the notification. As @devhoffmann said, we will discuss that with our team and try to resolve this issue.

devhoffmann, 2020-01-06:
I checked the german and european trademark register and could not find an entry for HAL that targets any kind of software.
From my understanding, it should therefore be safe to use the name HAL.

swallat, 2020-01-06:
@devhoffmann I agree, I couldn't find anything either. I suggest we close the issue for now and reopen it if we run into any trouble.

closed 2020-01-06 by swallat