emsec/hal #261: Importing hal_gui breaks graph shaders

RenWal, 2020-03-20:
**Describe the bug**
After running `import hal_gui`, the gates in the graph lose their text and selection color. Nets lose their selection color and pins are no longer rendered.

**To Reproduce**
Steps to reproduce the behavior:
1. Open any netlist
2. Type `import hal_gui` or `from hal_gui import gui` into the Python console or the Python editor and execute
3. Click on any gate or module

**Expected behavior**
The shaders must not break.

**Screenshots**
[image]

RenWal, 2020-03-20:
@dertob 
@Slowpoke42

RenWal, 2020-03-20:
Turned out to be some dependency issue on my side, sorry

closed 2020-03-20 by RenWal

nils1603, 2020-07-15:
[image] [View Manager](https://trello.com/c/2RZjVQbB/11-view-manager)