emsec/hal #390: Building Hal on Ubuntu 18.04

dahooz, 2021-04-21:
**Describe the bug**
Building Hal on Ubuntu 18.04 fails

## First error: missing `git describe`

**To Reproduce**
1. build as per documentation

**Expected behavior**
cmake running

**Observed behaviour**
cmake fails:

<details><summary>Error message</summary>
```console
$cmake ..
-- Selected build type: Release
-- ccache found!
-- System is:
-- Linux distro is: Ubuntu
-- Linux Distro version name is: bionic
-- System Arch is: x86_64
-- Extracting Version Information in PATH: /home/<omitted>/hal
ERROR: could not call 'git describe', result: 80e3640eb
VERBOSEVERSION_FILE: 3.2.6

-- HAL_VERSION:
CMake Error at CMakeLists.txt:81 (project):
  VERSION "OFF.OFF.OFF" format invalid.

-- Configuring incomplete, errors occurred!
```
</details>

**Workaround**
Define all the necessary variables with some dummy values:
```
cmake .. -DHAL_VERSION_MAJOR=0 -DHAL_VERSION_MINOR=1 -DHAL_VERSION_PATCH=1 -DHAL_VERSION_TWEAK=0 -DHAL_VERSION_ADDITIONAL_COMMITS=0 -DHAL_VERSION_DIRTY=false -DHAL_VERSION_BROKEN=false
```

To fix this properly, you would probably have to patch your [`cmake/hal_cmake_tools.cmake`](cmake/hal_cmake_tools.cmake) to have some sane defaults if `git describe` is not there.

## Second error (accessible after previous workaround applied) regarding `std::filesystem`

**Expected behavior**
make running

**Observed behaviour**
make fails:

<details><summary>cmake excerpt</summary>
Note that cmake does notice that ubuntu18.04 only has experimental filesystem available:

```console
[...]
-- Looking for C++ include filesystem
-- Looking for C++ include filesystem - not found
-- Looking for C++ include experimental/filesystem
-- Looking for C++ include experimental/filesystem - found
[...]
```

This is because [`cmake/FindFilesystem.cmake`](cmake/FindFilesystem.cmake) finds that filesystem is only available in experimental. It even sets a `CXX_FILESYSTEM_NAMESPACE` cmake variable that you could use as a preprocessor variable in the c++ code and replace every `std::filesystem` with `CXX_FILESYSTEM_NAMESPACE`.
</details>

<details><summary>Error message</summary>
```console
$make
[...]
In file included from /home/<omitted>/hal/include/hal_core/utilities/log.h:30:0,
                 from /home/<omitted>/hal/src/utilities/log.cpp:1:
/home/<omitted>/hal/include/hal_core/utilities/utils.h:475:66: error: ‘filesystem’ in namespace ‘std’ does not name a type
         CORE_API bool folder_exists_and_is_accessible(const std::filesystem::path& path);
[...]
```
</details>

dahooz, 2021-04-23:
## Fourth error ([since a very recent merge of the module dialog branch](https://github.com/emsec/hal/commit/0700e89a3514e12ce8d6581fe416d74417fce092)):

**Expected behavior**
make running

**Observed behaviour**
make fails:

> **Note:** cmake does not test the version of `qt`

<details><summary>Error message</summary>

```console
$ make
[...]
/home/<omitted>/hal/plugins/gui/src/module_dialog/module_select_model.cpp: In member function ‘void hal::ModuleSelectProxy::searchTextChanged(const QString&)’:
/home/<omitted>/hal/plugins/gui/src/module_dialog/module_select_model.cpp:174:9: error: ‘setFilterRegularExpression’ was not declared in this scope
         setFilterRegularExpression(txt);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/<omitted>/hal/plugins/gui/src/module_dialog/module_select_model.cpp:174:9: note: suggested alternative: ‘QRegularExpression’
         setFilterRegularExpression(txt);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
         QRegularExpression
plugins/gui/CMakeFiles/gui.dir/build.make:2436: recipe for target 'plugins/gui/CMakeFiles/gui.dir/src/module_dialog/module_select_model.cpp.o' failed
[...]
```
</details>

Explanation: qt5 ubuntu 18 is version [`5.9.5`](https://packages.ubuntu.com/bionic/qt5-default), while ubuntu 20 has [`5.12.8`](https://packages.ubuntu.com/bionic/qt5-default). For `QSortFilterProxyModel::setFilterRegularExpression` to exist, we need at least [`Qt 5.12`](https://doc.qt.io/qt-5/qsortfilterproxymodel.html#setFilterRegularExpression). You could add a check in cmake for qt5 version?

**Workaround**
install qt5.12.8 from a ppa:
```
sudo add-apt-repository ppa:beineri/opt-qt-5.12.8-bionic
sudo apt update
sudo apt install qt512svg qt512base
source /opt/qt512/bin/qt512-env.sh
rm -r build # and so on...
```

**Now hal builds again (hooray)**

joern274, 2021-04-25:
Current hal (master) software has been developed and compiled against libraries shipped with Ubuntu 20.04 distribution. A lot of required functionality was not yet available in  Ubuntu 18.04.

However - I created a branch branch **_backport/ubuntu_18_04_**  with a lot of code fixes so that hal compiles and runs on Ubuntu 18.04. once the required dependencies are installed. Disclaimer:
- z3 functionality not fully tested
- don't expect automatic tests to run
- don't expect future functionality to be back-ported as well

closed 2021-12-01 by SJulianS