emsec/hal #432: FindPythonInterp is deprecated, use FindPython3 instead

RenWal, 2021-10-14:
The following line causes a cmake failure on my Ubuntu 20.04 machine:
https://github.com/emsec/hal/blob/5e004f6010a9e690b56fe02b54f38a970f8fa5f9/cmake/detect_dependencies.cmake#L148

"Dual stack" python installs, i.e. having both python2 and python3 installed, will trigger this if the system's default python points to python2.

For this reason, as of cmake 3.12, FindPython3 should be used instead of FindPython.

RenWal, 2021-10-14:
(The same goes for `deps/pybind11` which still has the deprecated CMakeLists)

suzizecat, 2021-11-26:
As someone who just had the issue, one can work around this by using the cmake option 
```
-DPYTHON_EXECUTABLE=`which python3` 
```

nils1603, 2021-12-01:
@RenWal could you prepare a Pull Request?

RenWal, 2021-12-04:
Can someone from your team point out whether there's anything that prevents updating pybind to a version that doesn't use deprecated CMake?

nils1603, 2022-01-10:
The pybind version has been updated to 2.9. We also fixed the python3 include if both python2.7 and python3 are installed. This will be merged into the master with the new HAL v4.0

https://github.com/emsec/hal/tree/v4.0.0

RenWal, 2022-10-17:
This has been merged, closing.

closed 2022-10-17 by RenWal