emsec/hal #441: HAL Fails to parse SKY130 standard cell liberty files

suzizecat, 2021-11-26:
**Describe the bug**
Trying to open a .lib file from SKY130 PDK (https://github.com/google/skywater-pdk) fails with the following log in terminal.
For information the exact file is [sky130_fd_sc_ls__tt_100C_1v80.lib (zipped)](https://github.com/emsec/hal/files/7610702/sky130_fd_sc_ls__tt_100C_1v80.zip) 

```
[26.11.2021 22:06:46] [gui] [info] Trying to use gate library /home/julien/Projets/HDL/Hamster/synth/tech.lib.
[26.11.2021 22:06:46] [gate_library_parser] [info] selected gate library parser 'Default Liberty Parser'.
[26.11.2021 22:06:47] [liberty_parser] [warning] [/mnt/ramdisk/hal/plugins/liberty_parser/src/liberty_parser.cpp:529] unsupported pg_type 'nwell' of pg_pin 'VNB' for cell 'sky130_fd_sc_ls__a2111o_1'.
[26.11.2021 22:06:47] [liberty_parser] [warning] [/mnt/ramdisk/hal/plugins/liberty_parser/src/liberty_parser.cpp:529] unsupported pg_type 'pwell' of pg_pin 'VPB' for cell 'sky130_fd_sc_ls__a2111o_1'.
[26.11.2021 22:06:47] [liberty_parser] [error] [/mnt/ramdisk/hal/plugins/liberty_parser/src/liberty_parser.cpp:48] expected Token ';' but got 'false' near line 259.
[26.11.2021 22:06:47] [gate_library_parser] [error] [/mnt/ramdisk/hal/src/netlist/gate_library/gate_library_parser/gate_library_parser_manager.cpp:90] failed to parse gate library from file '/home/julien/Projets/HDL/Hamster/synth/tech.lib'.
[26.11.2021 22:06:47] [netlist] [critical] [/mnt/ramdisk/hal/src/netlist/netlist_factory.cpp:39] could not read netlist without gate library.
[26.11.2021 22:06:47] [gui] [error] [/mnt/ramdisk/hal/plugins/gui/src/file_manager/file_manager.cpp:288] Failed using gate library /home/julien/Projets/HDL/Hamster/synth/tech.lib.
[26.11.2021 22:06:47] [gui] [info] Searching for (other) compatible netlists.
[26.11.2021 22:06:47] [netlist_parser] [error] [/mnt/ramdisk/hal/src/netlist/netlist_parser/netlist_parser_manager.cpp:34] no hdl parser registered for file type '.lib'
[26.11.2021 22:06:47] [gui] [error] [/mnt/ramdisk/hal/plugins/gui/src/file_manager/file_manager.cpp:301] Unable to find a compatible gate library. Deserialization failed!
```

**To Reproduce**
Steps to reproduce the behavior:
1. Open HAL
2. Drop, once dezipped,  [sky130_fd_sc_ls__tt_100C_1v80.lib](https://github.com/emsec/hal/files/7610702/sky130_fd_sc_ls__tt_100C_1v80.zip) to the `OPEN FILE` part of the window
3. See error

**Expected behavior**
I would expect HAL to, at least partially, load the liberty file.

**Screenshots**

N/A.

**Desktop (please complete the following information):**
- OS: Ubuntu 20.04
- Version Built from sources, with all plugins, from master f987598

**Additional context**
This liberty file loads up correctly in various tools so far (Yosys, OpenSTA) 

Best regards !

SJulianS, 2021-12-01:
Thanks for the report! I just pushed a fix for this issue, at least on my machine HAL can now parse the file you provided. Please check again using the current master.

suzizecat, 2021-12-01:
Thanks for the fix. However, it seems that you're not quite there yet.
On 4c1b66657c69ef14da288f539e11081609451f8e have: 
```
[01.12.2021 20:37:29] [core] [info] setting log file to '/home/julien/Data/Linux/OpenLane/pdks/skywater-pdk/libraries/sky130_fd_sc_ls/latest/timing/sky130_fd_sc_ls__tt_100C_1v80.log'.
[01.12.2021 20:37:29] [gui] [info] Trying to use gate library /home/julien/Data/Linux/OpenLane/pdks/skywater-pdk/libraries/sky130_fd_sc_ls/latest/timing/sky130_fd_sc_ls__tt_100C_1v80.lib.
[01.12.2021 20:37:29] [gate_library_parser] [info] selected gate library parser 'Default Liberty Parser'.
[01.12.2021 20:37:30] [gate_library_parser] [info] parsed gate library 'sky130_fd_sc_ls__tt_100C_1v80' from file '/home/julien/Data/Linux/OpenLane/pdks/skywater-pdk/libraries/sky130_fd_sc_ls/latest/timing/sky130_fd_sc_ls__tt_100C_1v80.lib' in 0.79 seconds.
[01.12.2021 20:37:30] [gate_library_manager] [info] gate library did not contain a GND gate, auto-generated type 'HAL_GND'.
[01.12.2021 20:37:30] [gate_library_manager] [info] gate library did not contain a VDD gate, auto-generated type 'HAL_VDD'.
[01.12.2021 20:37:30] [netlist_parser] [error] [/mnt/ramdisk/hal/src/netlist/netlist_parser/netlist_parser_manager.cpp:34] no hdl parser registered for file type '.lib'
[01.12.2021 20:37:30] [gui] [error] [/mnt/ramdisk/hal/plugins/gui/src/file_manager/file_manager.cpp:288] Failed using gate library /home/julien/Data/Linux/OpenLane/pdks/skywater-pdk/libraries/sky130_fd_sc_ls/latest/timing/sky130_fd_sc_ls__tt_100C_1v80.lib.
[01.12.2021 20:37:30] [gui] [info] Searching for (other) compatible netlists.
[01.12.2021 20:37:30] [netlist_parser] [error] [/mnt/ramdisk/hal/src/netlist/netlist_parser/netlist_parser_manager.cpp:34] no hdl parser registered for file type '.lib'
[01.12.2021 20:37:30] [gui] [error] [/mnt/ramdisk/hal/plugins/gui/src/file_manager/file_manager.cpp:301] Unable to find a compatible gate library. Deserialization failed!
``` 

So it looks like you have properly read the file but at the end the parser is not registered for further processing or something.

SJulianS, 2021-12-04:
It looks like you are trying to open a liberty gate library file as a netlist:
`[01.12.2021 20:37:30] [netlist_parser] [error] [/mnt/ramdisk/hal/src/netlist/netlist_parser/netlist_parser_manager.cpp:34] no hdl parser registered for file type '.lib'`

Please try again with a Verilog or VHDL netlist synthesized for the SKY130 PDK. HAL will automatically select the appropriate gate library if it is placed in the correct folder (place it in `hal/plugins/gate_libraries/definitions` before building  HAL or `hal/build/share/hal/gate_libraries` afterwards). We're also currently working on changes to the handling of gate libraries when loading a new netlist file, which should clear things up in the future and hopefully prevent such mistakes :)

suzizecat, 2021-12-05:
Oooh ! That was indeed what i've done, since the pdk wasn't available in HAL, though that I needed to open the .lib to "convert it" to HAL format. 
I will try your flow which should indeed work way better.

closed 2021-12-06 by SJulianS