emsec/hal #452: Cannot compile hal, 'make' errors out

jarvis-lorry, 2022-03-02:
I'm trying to compile hal on Ubuntu 20.04.4 and I've gotten cmake to work but I when I run 'make', it errors out.  I don't understand the error.

Here is the output from make:

---@----vb:~/hal/build$ make
[  0%] Building CXX object src/utilities/CMakeFiles/utilities.dir/hal_file_manager.cpp.o
[  0%] Building CXX object src/utilities/CMakeFiles/utilities.dir/log.cpp.o
[  0%] Building CXX object src/utilities/CMakeFiles/utilities.dir/program_arguments.cpp.o
[  1%] Building CXX object src/utilities/CMakeFiles/utilities.dir/program_options.cpp.o
[  1%] Building CXX object src/utilities/CMakeFiles/utilities.dir/utils.cpp.o
[  1%] Linking CXX shared library ../../lib/libhal_utilities.so
/usr/bin/ld: /usr/local/lib/libspdlog.a(spdlog.cpp.o): relocation R_X86_64_TPOFF32 against `_ZGVZN6spdlog7details2os9thread_idEvE3tid' can not be used when making a shared object; recompile with -fPIC
collect2: error: ld returned 1 exit status
make[2]: *** [src/utilities/CMakeFiles/utilities.dir/build.make:162: lib/libhal_utilities.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:684: src/utilities/CMakeFiles/utilities.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

nils1603, 2022-03-02:
Hey @jarvis-lorry,

we are currently facing some version deprecations with some dependencies. HAL has not been updated to the most recent version of spdlog, which will happen with the new release.

You can either try the new v4.0.0 dev branch, but beware, there are many many many many many many things not working yet.

For now you can remove spdlog from your system (`sudo apt remove libspdlog-dev`) and let HAL build an older version itself.

nils1603, 2022-03-03:
Don't execute the install_dependencies again AFTER you deleted spdlog. You only need to execute the install dependencies script once.

jarvis-lorry, 2022-03-03:
I've tried what you suggested but still have the same error when I run 'make'.  This after I've removed spdlog.  I may have to try the development version, v4.0.0 dev.

nils1603, 2022-03-07:
I made some updates to the build process on the master. Let me know if this helps.

jarvis-lorry, 2022-03-07:
Should I use the latest commit?

nils1603, 2022-03-08:
Yes, use the latest commit on the master branch, so basically just pull and try again

jarvis-lorry, 2022-03-09:
Hi, I've been able to build HAL successfully and run it.  Now I have to learn how to use it.

SJulianS, 2022-03-10:
I'd recommend getting started with our Wiki: https://github.com/emsec/hal/wiki. In case you get stuck and need help, feel free to drop us a mail: hal@csp.mpg.de. HAL is still merely developed by researchers and we are always short on time, but we'll try to help as good as we possibly can.