emsec/hal #459: Verilog Parser occasionally incorrectly parses signal names

reillymck, 2022-04-29:
It seems that occasionally the verilog parser does not properly parse escaped identifiers. It appears to happen specifically when the escaped identifier is used as in i/o when instantiating a module. The problem is made noticeable when a double slash, "//", is in the escaped signal name.

For example: 
```
LUT6 #(
    .INIT(64'h0000000000004447)) 
    \or1200_cfgr//i_ 
       (.I0(du_addr_IBUF[2]),
        .I1(\or1200_sprs/du_access ),
        .I2(operand_a[2]),
        .I3(ex_insn_OBUF[2]),
        .I4(spr_addr_OBUF[1]),
        .I5(spr_addr_OBUF[3]),
        .O(\or1200_cfgr//i__n_0 ));
```

The signal \or1200_cfgr//i__n_0 is parsed incorrectly. The "//" is interpreted by the parser as a comment. Interestingly enough, the signal \or1200_cfgr//i_  is parsed correctly. 

When trying to load this file as a netlist in hal, the verilog parser provides the following error message: 
```
[netlist_parser] [info] parsing '/home/reilly/vtr_benchmarks/results/or1200/or1200.v'...
[verilog_parser] [warning] [/home/reilly/RegAnalysis/hal/plugins/verilog_parser/src/verilog_parser.cpp:419] cannot parse compiler directives.
[verilog_parser] [error] [/home/reilly/RegAnalysis/hal/plugins/verilog_parser/src
[or1200.txt](https://github.com/emsec/hal/files/8588279/or1200.txt)
/verilog_parser.cpp:51] expected Token ')' but got 'or1200_cfgr/spr_dat_o_reg[0]' near line 17034.
[netlist_parser] [error] [/home/reilly/RegAnalysis/hal/src/netlist/netlist_parser/netlist_parser_manager.cpp:47] could not parse file '/home/reilly/vtr_benchmarks/results/or1200/or1200.v'.
```

I have included the full verilog file that this fails on. This verilog file was generated post-synthesis using the vivado tcl `write_verilog` command. Note that github does not support the .v filetype, so I have uploaded it as .txt

reillymck, 2022-04-29:
[or1200.txt](https://github.com/emsec/hal/files/8588285/or1200.txt)

SJulianS, 2022-05-02:
thanks for the report, this is now fixed within the v4.0.0 branch

closed 2022-05-02 by SJulianS