emsec/hal #472: Issue while launching HAL

Guttybitty, 2022-10-02:
After install HAL, when hal -g or any command is run, it is giving following error. 

hal: error while loading shared libraries: libhal_plugin_system.so: cannot open shared object file: No such file or directory 
I have attached logs while installation. Please let me know what the issue is.
[log.txt](https://github.com/emsec/hal/files/9692340/log.txt)

joern274, 2022-10-05:
Thanks for submitting the log.

I noticed that (after successfully building hal in build directory) the first attempt to launch hal was made by typing **hal -g**. Please try **bin/hal -g** instead and hal should start immediately in GUI mode.

I will look into the install problem but it might take some time.

thud06, 2022-10-14:
Hello, I am experiencing the same issue. bin/hal -g does not seem to work for me either.

joern274, 2022-10-15:
@thud06 please provide a log file like Guttybitty did to verify that the build process completed successfully or to narrow down problems during build. What was the response to bin/hal -g  ? What Operating system, distribution, and version are you using?

thud06, 2022-10-17:
@joern274 Here is the log file:

```
username@username-VirtualBox:~/Desktop/HAL/hal/build$ hal -g

Command 'hal' not found, did you mean:

  command 'khal' from deb khal (1:0.9.10-1.1)
  command 'al' from deb mono-devel (6.8.0.105+dfsg-2)
  command 'pal' from deb pal (0.4.3-8.1build4)
  command 'cal' from deb bsdmainutils (11.1.2ubuntu3)
  command 'fal' from deb dnet-progs (2.65build2)
  command 'halt' from deb systemd-sysv (245.4-4ubuntu3.18)
  command 'halt' from deb molly-guard (0.7.2)
  command 'haml' from deb ruby-haml (5.1.2-1)
  command 'hcal' from deb hdate (1.6.02-2ubuntu2)

Try: sudo apt install <deb name>
```

I am using Ubuntu 20.04.5 LTS. I am using HAL version 3.3.0. 

When I use bin/hal -g, I get the error

[bin/hal -g](bash: bin/hal: No such file or directory)

Thanks

joern274, 2022-10-17:
@thud06 from the information you provided we learn that most likely the hal build failed. Unfortunately does your log file not cover step 1.-5. from (["Building Hal"](https://github.com/emsec/hal/wiki/Building-HAL)). Can you run the build process again and send its output, please? Thanks.

thud06, 2022-10-18:
@joern274 I tried a fresh install, and the hal -g command still gave the same error. Here are the last few lines after running make

```
[100%] Linking CXX shared library ../../lib/hal_plugins/verilog_writer.so
[100%] Built target verilog_writer
Scanning dependencies of target vhdl_parser
[100%] Building CXX object plugins/vhdl_parser/CMakeFiles/vhdl_parser.dir/src/plugin_vhdl_parser.cpp.o
[100%] Building CXX object plugins/vhdl_parser/CMakeFiles/vhdl_parser.dir/src/vhdl_parser.cpp.o
[100%] Linking CXX shared library ../../lib/hal_plugins/vhdl_parser.so
[100%] Built target vhdl_parser
username@username-VirtualBox:~/Desktop/HAL2/hal/build$ hal -g

Command 'hal' not found, did you mean:

  command 'al' from deb mono-devel (6.8.0.105+dfsg-2)
  command 'halt' from deb systemd-sysv (245.4-4ubuntu3.18)
  command 'halt' from deb molly-guard (0.7.2)
  command 'haml' from deb ruby-haml (5.1.2-1)
  command 'fal' from deb dnet-progs (2.65build2)
  command 'khal' from deb khal (1:0.9.10-1.1)
  command 'cal' from deb bsdmainutils (11.1.2ubuntu3)
  command 'hcal' from deb hdate (1.6.02-2ubuntu2)
  command 'pal' from deb pal (0.4.3-8.1build4)

Try: sudo apt install <deb name>

```

For the cmake step, I did not add any options, just ran it as "cmake .." Could this be the issue? If so, what should I add to it?

Thanks

joern274, 2022-10-18:
@thud06 Although the log does not cover the entire build process it ends with 100%, these are good news. Next I would expect the command **bin/hal -g** as the executable is supposed to be located in the **bin** directory.

- can you verify that the **bin** directory was created?
- can you list the content of that directory?

Running **cmake ..** without any options is fine to get hal up and running. Popular options are **-DBUILD_ALL_PLUGINS=1** to build all the powerful plugins like DANA and Simulator or **-DBUILD_DOCUMENTATION=1** to build documentation as well. But let's make sure first that the build you issued was successful and that you can run the software.

closed 2022-10-18 by joern274