emsec/hal #473: HAL installing issues

tanvir-jewel, 2022-10-07:
```
-- Performing Test supports_-fno-optimize-sibling-calls - Success
-- HAL_STUDY mode turned OFF!
-- Could not find z3
-- GATE_LIBS: /home/t177h608/My_Files/HAL/hal/plugins/gate_libraries/definitions/.gitignore;/home/t177h608/My_Files/HAL/hal/plugins/gate_libraries/definitions/NanGate_15nm_OCL.hgl;/home/t177h608/My_Files/HAL/hal/plugins/gate_libraries/definitions/NangateOpenCellLibrary.hgl;/home/t177h608/My_Files/HAL/hal/plugins/gate_libraries/definitions/XILINX_SIMPRIM.hgl;/home/t177h608/My_Files/HAL/hal/plugins/gate_libraries/definitions/XILINX_UNISIM.hgl;/home/t177h608/My_Files/HAL/hal/plugins/gate_libraries/definitions/example_library.hgl;/home/t177h608/My_Files/HAL/hal/plugins/gate_libraries/definitions/ice40ultra.hgl;/home/t177h608/My_Files/HAL/hal/plugins/gate_libraries/definitions/lsi_10k.hgl
-- Unable to find IGRAPH header files!
-- Unable to find IGRAPH library files!
-- CPACK_GENERATOR: 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
/home/t177h608/My_Files/HAL/hal/src/netlist/Z3_INCLUDE_DIRS
   used as include directory in directory /home/t177h608/My_Files/HAL/hal/src/netlist
IGRAPH_LIBRARIES (ADVANCED)
    linked by target "graph_algorithm" in directory /home/t177h608/My_Files/HAL/hal/plugins/graph_algorithm
Z3_LIBRARIES
    linked by target "netlist" in directory /home/t177h608/My_Files/HAL/hal/src/netlist
    linked by target "solve_fsm" in directory /home/t177h608/My_Files/HAL/hal/plugins/solve_fsm
    linked by target "z3_utils" in directory /home/t177h608/My_Files/HAL/hal/plugins/z3_utils

-- Configuring incomplete, errors occurred!
See also "/home/t177h608/My_Files/HAL/hal/build/CMakeFiles/CMakeOutput.log".
See also "/home/t177h608/My_Files/HAL/hal/build/CMakeFiles/CMakeError.log".

```
I am facing this issues while trying to build it with cmake. What would be the possible solution for this?

joern274, 2022-10-07:
While the reason for failed installation seems to be obvious (missing IGRAPH header and library) some more information is needed to reproduce the problem.
1. Operation system distribution and version (Ubuntu?)
2. Did you run the install_dependencies.sh script?
3. Any error messages while running the script above?
To solve the installation problem it might be good enough to install the developer version of IGRAPH library <= 0.9.10 manually on your system.

joern274, 2022-10-17:
@tanvir-jewel have you been able to solve your problem? I would like to close this issue unless you could provide little more information.

closed 2022-11-02 by SJulianS