emsec/hal #477: Segmentation fault when creating new netlist from GUI

leongross, 2022-10-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. open hal gui `./hal -g`
2. 
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OS: [e.g. iOS]
- Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

RenWal, 2022-10-20:
Looks like the gate libraries aren't loaded before showing that dialog. I think this was introduced in c78aac094fa89cd953746a12e9f563bae5c89f7a in the process of moving to our new project manager and import system.

While we work on a full solution, I've prepared a hotfix in [branch fix/create-netlist-no-libraries](https://github.com/emsec/hal/tree/fix/create-netlist-no-libraries).

Looking at the code in [/plugins/gui/src/main_window/main_window.cpp#L581](https://github.com/emsec/hal/blob/master/plugins/gui/src/main_window/main_window.cpp#L581), it looks like this might need some additional work. Currently, this is using neither the User Action System nor the FileManager. @SJulianS @joern274 should we consider updating that method?

RenWal, 2022-10-20:
Stumbled upon a few more things:

- netlists opened this way can't be closed (nothing happens)
- the "save" button is clickable but does nothing, the same goes for the keybind
- simulator temp files end up in the working directory (we call `ProjectManager::instance()->get_project_directory()` without checking whether a project is currently open, not sure if that is OK)
- the log isn't saved

We can probably simplify some stuff and get rid of those special cases if we enforce that a project is always created when we create a netlist – could be a temporary directory, or maybe even just immediately trigger a save dialog when creating a new netlist?

joern274, 2022-10-21:
Please check whether issue has been fixed by now.

RenWal, 2022-10-24:
Not crashing anymore, and I like the new dialog.

One minor thing, and certainly personal preference: Would it make sense to only allow the user to proceed after selecting a gate library? Right now, if you don't select one, you can still confirm the dialog and then you receive an error message.

joern274, 2022-10-25:
Good point and easy to accomplish

closed 2022-11-02 by SJulianS