emsec/hal #480: Several crashes related to inaccessible files

RenWal, 2022-10-24:
While testing the new netlist import logic, I've found some bugs related to handling files that are not accessible when reading or writing.

- [ ] When attempting to create a project at a location that HAL can't write to (read-only, or permission denied), HAL crashes.
- [ ] When a project directory disappears while HAL is running, HAL crashes when attempting to (auto-)save the project. This one might be relevant in case your network-attached storage goes away briefly while you're working and HAL tries to auto-save. Interestingly, making the project directory read-only does lead to a sensible error message.
- [ ] When starting HAL, the "recent projects" screen shows all projects that still exist on disk. When we now delete one of these projects from disk and then click them in the recent projects screen, HAL crashes. No such crash occurs when we simply remove read permissions, the issue seems to be specifically triggered by deletion.
- [ ] When adding a file suffix on a project directory when creating a new project, or saving the project to a new location, HAL will silently cut off that suffix before creating the project directory on disk.
- [ ] Putting a file suffix on a project directory (like `project.old`) manually using a file manager will lead to a crash when opening that project, due to the suffix being stripped off somewhere during processing, causing HAL to attempt to open a non-existing directory.
- [ ] Attempting to "save-as" a Python script to a location without write permission will cause an error message in the log, but the editor's internal file path will still update. Clicking "save" will then still try to write to that read-only location. I'd argue this is personal preference, but I would expect the file path to not update in that case.
- [ ] Attempting to save a macro to a read-only location is silently ignored, there is no error message or log entry

joern274, 2022-10-25:
@RenWal thanks for thorough testing and reporting all the bugs. However, TMHO we will stick to the strategy that project directories in HAL must not have a suffix and any given suffix will be chopped. The suffix (or not having a suffix) is used by project management tools as well as in Qt::QFileDialog to differentiate and filter between files we can import (.vhdl, .v, .hal) and projects we might be able to open.
There are more tests to follow to identify a valid HAL project but these tests are more expansive, so I would not like to see them executed for every single directory entry regardless whether it has a suffix or not.

joern274, 2022-11-09:
According to previous comment project directories with suffix in name will not be supported. Fix to "Putting a file suffix on a project directory ..." : HAL will tell user immediately that project cannot be opened because of suffix. No more crashes :+1:

RenWal, 2022-11-17:
Nice! Did some tests and found two minor remaining issues, I think all other points are well addressed now.

Test results:

- Regarding point 4: importing a netlist into a new project directory called `foo.bar` will result in the project actually being saved to `foo`, but `foo.bar` is inserted into the recent files overview (and immediately declared missing). To be consistent, we should also reject such directory names when importing, just like we now do in the save-as dialog.
- Regarding point 7: this now works, however, it's not clear to me what the difference between the buttons "close without saving" and "cancel" is, could we make this more explicit?
[image]

joern274, 2022-11-18:
@RenWal thanks for testing.

Fixed "Regarding point 4" by disallowing suffix in chosen project name as suggested.

"Regarding point 7": Although it is understandable that users want to have more explicit hints how to deal with errors we should stick to the standard procedures, icons and buttons the system is providing. The layout of the buttons in the screen shot above is the result of using Qt standard buttons Discard (stop recording, don't save macro) and Cancel (oops, click on stop recording was a mistake). Using Qt standard buttons is a must when porting the application to different platforms. The result is not very explicit -agreed- but should look familiar to the user of the system.

RenWal, 2022-11-18:
I'm with you on keeping the standard buttons. Still, it would not have been clear to me (without looking into the source code) what the difference between Discard and Cancel is – especially, I would not have expected recording to continue after clicking Cancel (though I think it's a good idea to have that option).

Would it be an option to put some remark into the dialog text, and then keep the buttons as-is?

closed 2023-01-13 by joern274