emsec/hal #481: Can not read ISCAS 85 Benchmarks

thud06, 2022-10-25:
I am trying to analyze the ISCAS 85 benchmarks particularly c432.v, and am receiving the error : 

```
[25.10.2022 12:58:08] [netlist_parser] [error] [/home/user/Desktop/HAL/hal/src/netlist/netlist_parser/netlist_parser_manager.cpp:47] failed to parse '/home/user/Desktop/c432.v':
[/home/user/Desktop/HAL/hal/plugins/verilog_parser/src/verilog_parser.cpp:477] could not parse Verilog file '/home/user/Desktop/c432.v': expected Token ')' but got 'G118' (line 20)
[25.10.2022 12:58:08] [gui] [error] [/home/user/Desktop/HAL/hal/plugins/gui/src/file_manager/file_manager.cpp:443] Unable to find a compatible gate library. Deserialization failed!
```

I've tried making my own library, but I am still receiving the same error.

The c432.v file is:

```

module c432(G1,G10,G11,G12,G13,G14,G15,G16,G17,G18,G19,G2,G20,G21,G22,G23,
  G24,G25,G26,G27,G28,G29,G3,G30,G31,G32,G33,G34,G35,G36,G4,G426,G427,G428,
  G429,G430,G431,G432,G5,G6,G7,G8,G9);
input G1,G2,G3,G4,G5,G6,G7,G8,G9,G10,G11,G12,G13,G14,G15,G16,G17,G18,G19,G20,
  G21,G22,G23,G24,G25,G26,G27,G28,G29,G30,G31,G32,G33,G34,G35,G36;
output G426,G427,G428,G429,G430,G431,G432;

  wire G118,G119,G122,G123,G126,G127,G130,G131,G134,G135,G138,G139,G142,G143,
    G146,G147,G150,G151,G154,G157,G158,G159,G162,G165,G168,G171,G174,G177,G180,
    G183,G184,G185,G186,G187,G188,G189,G190,G191,G192,G193,G194,G195,G196,G197,
    G198,G199,G203,G213,G223,G226,G229,G232,G235,G238,G241,G242,G245,G246,G249,
    G250,G253,G254,G255,G256,G257,G258,G259,G262,G263,G266,G269,G272,G275,G278,
    G281,G284,G287,G288,G289,G290,G291,G292,G293,G294,G295,G299,G300,G301,G302,
    G303,G304,G305,G306,G307,G308,G318,G328,G329,G330,G331,G332,G333,G334,G335,
    G336,G337,G338,G339,G340,G341,G342,G343,G344,G345,G346,G347,G348,G349,G350,
    G351,G352,G353,G354,G355,G358,G368,G369,G370,G371,G372,G373,G374,G375,G376,
    G377,G378,G383,G390,G396,G401,G404,G408,G411,G412,G413,G414,G415,G416,G417,
    G418,G421,G424,G425;

  not NOT_0(G118,G1);
  not NOT_1(G119,G2);
  not NOT_2(G122,G4);
  not NOT_3(G123,G6);
  not NOT_4(G126,G8);
  not NOT_5(G127,G10);
  not NOT_6(G130,G12);
  not NOT_7(G131,G14);
  not NOT_8(G134,G16);
  not NOT_9(G135,G18);
  not NOT_10(G138,G20);
  not NOT_11(G139,G22);
  not NOT_12(G142,G24);
  not NOT_13(G143,G26);
  not NOT_14(G146,G28);
  not NOT_15(G147,G30);
  not NOT_16(G150,G32);
  not NOT_17(G151,G34);
  nand NAND2_0(G154,G118,G2);
  nor NOR2_0(G157,G3,G119);
  nor NOR2_1(G158,G5,G119);
  nand NAND2_1(G159,G122,G6);
  nand NAND2_2(G162,G126,G10);
  nand NAND2_3(G165,G130,G14);
  nand NAND2_4(G168,G134,G18);
  nand NAND2_5(G171,G138,G22);
  nand NAND2_6(G174,G142,G26);
  nand NAND2_7(G177,G146,G30);
  nand NAND2_8(G180,G150,G34);
  nor NOR2_2(G183,G7,G123);
  nor NOR2_3(G184,G9,G123);
  nor NOR2_4(G185,G11,G127);
  nor NOR2_5(G186,G13,G127);
  nor NOR2_6(G187,G15,G131);
  nor NOR2_7(G188,G17,G131);
  nor NOR2_8(G189,G19,G135);
  nor NOR2_9(G190,G21,G135);
  nor NOR2_10(G191,G23,G139);
  nor NOR2_11(G192,G25,G139);
  nor NOR2_12(G193,G27,G143);
  nor NOR2_13(G194,G29,G143);
  nor NOR2_14(G195,G31,G147);
  nor NOR2_15(G196,G33,G147);
  nor NOR2_16(G197,G35,G151);
  nor NOR2_17(G198,G36,G151);
  and AND9_0(G199,G154,G159,G162,G165,G168,G171,G174,G177,G180);
  not NOT_18(G203,G199);
  not NOT_19(G213,G199);
  xor XOR2_0(G223,G203,G154);
  xor XOR2_1(G226,G203,G159);
  xor XOR2_2(G229,G203,G162);
  xor XOR2_3(G232,G203,G165);
  xor XOR2_4(G235,G203,G168);
  xor XOR2_5(G238,G203,G171);
  nand NAND2_9(G241,G1,G213);
  xor XOR2_6(G242,G203,G174);
  nand NAND2_10(G245,G213,G4);
  xor XOR2_7(G246,G203,G177);
  nand NAND2_11(G249,G213,G8);
  xor XOR2_8(G250,G203,G180);
  nand NAND2_12(G253,G213,G12);
  nand NAND2_13(G254,G213,G16);
  nand NAND2_14(G255,G213,G20);
  nand NAND2_15(G256,G213,G24);
  nand NAND2_16(G257,G213,G28);
  nand NAND2_17(G258,G213,G32);
  nand NAND2_18(G259,G223,G157);
  nand NAND2_19(G262,G223,G158);
  nand NAND2_20(G263,G226,G183);
  nand NAND2_21(G266,G229,G185);
  nand NAND2_22(G269,G232,G187);
  nand NAND2_23(G272,G235,G189);
  nand NAND2_24(G275,G238,G191);
  nand NAND2_25(G278,G242,G193);
  nand NAND2_26(G281,G246,G195);
  nand NAND2_27(G284,G250,G197);
  nand NAND2_28(G287,G226,G184);
  nand NAND2_29(G288,G229,G186);
  nand NAND2_30(G289,G232,G188);
  nand NAND2_31(G290,G235,G190);
  nand NAND2_32(G291,G238,G192);
  nand NAND2_33(G292,G242,G194);
  nand NAND2_34(G293,G246,G196);
  nand NAND2_35(G294,G250,G198);
  and AND9_1(G295,G259,G263,G266,G269,G272,G275,G278,G281,G284);
  not NOT_20(G299,G262);
  not NOT_21(G300,G287);
  not NOT_22(G301,G288);
  not NOT_23(G302,G289);
  not NOT_24(G303,G290);
  not NOT_25(G304,G291);
  not NOT_26(G305,G292);
  not NOT_27(G306,G293);
  not NOT_28(G307,G294);
  not NOT_29(G308,G295);
  not NOT_30(G318,G295);
  xor XOR2_9(G328,G308,G259);
  xor XOR2_10(G329,G308,G263);
  xor XOR2_11(G330,G308,G266);
  xor XOR2_12(G331,G308,G269);
  nand NAND2_36(G332,G3,G318);
  xor XOR2_13(G333,G308,G272);
  nand NAND2_37(G334,G318,G7);
  xor XOR2_14(G335,G308,G275);
  nand NAND2_38(G336,G318,G11);
  xor XOR2_15(G337,G308,G278);
  nand NAND2_39(G338,G318,G15);
  xor XOR2_16(G339,G308,G281);
  nand NAND2_40(G340,G318,G19);
  xor XOR2_17(G341,G308,G284);
  nand NAND2_41(G342,G318,G23);
  nand NAND2_42(G343,G318,G27);
  nand NAND2_43(G344,G318,G31);
  nand NAND2_44(G345,G318,G35);
  nand NAND2_45(G346,G328,G299);
  nand NAND2_46(G347,G329,G300);
  nand NAND2_47(G348,G330,G301);
  nand NAND2_48(G349,G331,G302);
  nand NAND2_49(G350,G333,G303);
  nand NAND2_50(G351,G335,G304);
  nand NAND2_51(G352,G337,G305);
  nand NAND2_52(G353,G339,G306);
  nand NAND2_53(G354,G341,G307);
  and AND9_2(G355,G346,G347,G348,G349,G350,G351,G352,G353,G354);
  not NOT_31(G358,G355);
  nand NAND2_54(G368,G5,G358);
  nand NAND2_55(G369,G358,G9);
  nand NAND2_56(G370,G358,G13);
  nand NAND2_57(G371,G358,G17);
  nand NAND2_58(G372,G358,G21);
  nand NAND2_59(G373,G358,G25);
  nand NAND2_60(G374,G358,G29);
  nand NAND2_61(G375,G358,G33);
  nand NAND2_62(G376,G358,G36);
  nand NAND4_0(G377,G2,G241,G332,G368);
  nand NAND4_1(G378,G245,G334,G369,G6);
  nand NAND4_2(G383,G249,G336,G370,G10);
  nand NAND4_3(G390,G253,G338,G371,G14);
  nand NAND4_4(G396,G254,G340,G372,G18);
  nand NAND4_5(G401,G255,G342,G373,G22);
  nand NAND4_6(G404,G256,G343,G374,G26);
  nand NAND4_7(G408,G257,G344,G375,G30);
  nand NAND4_8(G411,G258,G345,G376,G34);
  not NOT_32(G412,G377);
  and AND8_0(G413,G378,G383,G390,G396,G401,G404,G408,G411);
  not NOT_33(G414,G390);
  not NOT_34(G415,G401);
  not NOT_35(G416,G404);
  not NOT_36(G417,G408);
  nand NAND2_63(G418,G383,G414);
  nand NAND4_9(G421,G383,G390,G415,G396);
  nand NAND3_0(G424,G396,G390,G416);
  nand NAND4_10(G425,G383,G390,G404,G417);
  not NOT_37(G426,G199);
  not NOT_38(G427,G295);
  not NOT_39(G428,G355);
  nor NOR2_18(G429,G412,G413);
  nand NAND4_11(G430,G378,G383,G418,G396);
  nand NAND4_12(G431,G378,G383,G421,G424);
  nand NAND4_13(G432,G378,G418,G421,G425);

endmodule
```

- OS: Ubuntu 20.01
- Version hal 3.3.0

thud06, 2022-10-26:
@SJulianS I am looking forward to this addition. Also thanks for the hard work you are putting into the project !

SJulianS, 2022-10-29:
Please check again with branch [branch feature/parser_update](https://github.com/emsec/hal/tree/feature/parser_update), that should do the trick. If netlist instantiation still fails, please provide the respective gate library as well :)

closed 2022-10-31 by SJulianS

thud06, 2022-11-01:
> Please check again with branch [branch feature/parser_update](https://github.com/emsec/hal/tree/feature/parser_update), that should do the trick. If netlist instantiation still fails, please provide the respective gate library as well :)

Hello, I cant seem to find the branch in https://github.com/emsec/hal/branches/active, And for some reason your posted link isn't working

RenWal, 2022-11-05:
The branch has been deleted in the meantime, since #483 has been merged. Please test the current master.