emsec/hal #490: error: cannot convert ‘igraph_matrix_t*’ to ‘igraph_matrix_int_t*

leongross, 2022-11-27:
**Describe the bug**
Compiling hal with all it's plugins fails when compiling the `communities_fast_greedy` plugin.

**To Reproduce**
Steps to reproduce the behavior:
1. `git clone https://github.com/emsec/hal.git && cd hal`
1. `./install_dependencies.sh` 
1. `mkdir build && cd build`
1. `cmake -DBUILD_ALL_PLUGINS=1 ..`
1. `make`

**Error log**
```
~/hal/plugins/graph_algorithm/src/clustering/communities_fast_greedy.cpp: In member function ‘std::map<int, std::set<hal::Gate*> > hal::GraphAlgorithmPlugin::get_communities_fast_greedy(hal::Netlist*)’:
~/hal/plugins/graph_algorithm/src/clustering/communities_fast_greedy.cpp:35:37: error: cannot convert ‘igraph_matrix_t*’ to ‘igraph_matrix_int_t*’
   35 |                                     &merges,
      |                                     ^~~~~~~
      |                                     |
      |                                     igraph_matrix_t*
In file included from /usr/include/igraph/igraph.h:72,
                 from ~/hal/plugins/graph_algorithm/include/graph_algorithm/plugin_graph_algorithm.h:30,
                 from ~/hal/plugins/graph_algorithm/src/clustering/communities_fast_greedy.cpp:1:
/usr/include/igraph/igraph_community.h:122:68: note:   initializing argument 3 of ‘igraph_error_t igraph_community_fastgreedy(const igraph_t*, const igraph_vector_t*, igraph_matrix_int_t*, igraph_vector_t*, igraph_vector_int_t*)’
  122 |                                               igraph_matrix_int_t *merges,
      |                                               ~~~~~~~~~~~~~~~~~~~~~^~~~~~
make[2]: *** [plugins/graph_algorithm/CMakeFiles/graph_algorithm.dir/build.make:76: plugins/graph_algorithm/CMakeFiles/graph_algorithm.dir/src/clustering/communities_fast_greedy.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1066: plugins/graph_algorithm/CMakeFiles/graph_algorithm.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

```

**Expected behavior**
The plugin should compile.

**Desktop (please complete the following information):**
- OS: Arch Linux `6.0.9-arch1-1`
- Version: commit `e63298d7124b0b5906f50dcca4e6ef576a6d6822`
- igraph: `0.10.2-1`

RenWal, 2022-11-27:
related https://github.com/emsec/hal/pull/487

joern274, 2022-11-28:
@leongross : Although @RenWal is right pointing out that future integration of IGRAPH into HAL is still disputed and on our todo list - however - the problem you mentioned should disappear if you compile and link HAL against IGRAPH version <= 0.9.x. From the error message you reported I guess that you have IGRAPH >= 0.10 installed on your system. That is where a major API break took place. 

An IGRAPH version compatible with HAL should be installed automatically when running `./install_dependencies`  on ubuntu 20.04 or 22.04 or MacOS. If you are using another distribution or if the automated procedure failed, please downgrade the IGRAPH library. if there is no binary package available for your distribution you might want to build it from [source](https://github.com/igraph/igraph/tree/release/0.9) .

BTW: which operating system / distribution are you using ?

szhorvat, 2022-11-28:
In case it helps, you can track which igraph version various package managers have: https://repology.org/project/igraph/versions

leongross, 2022-11-29:
Thanks for the answer. Yes, this seems to be an issue with `igraph`.

> OS: Arch Linux 6.0.9-arch1-1

I use Arch Linux which is (kind of?) supported, at least there is an entry in the dependency installer. In PR #493 I updated the dependency installer script to downgrade the `igraph` version.

closed 2024-06-06 by joern274