emsec/hal #520: make install from git fails

xiota, 2023-05-12:
**Describe the bug**
Running `cmake DESTDIR="/path/to/dest" install` fails with error:
```
CMake Error at cmake_install.cmake:93 (file):
  file INSTALL cannot find
  ".../hal-git/deps/spdlog-1.5.0": No such file or directory.
```

**To Reproduce**
Steps to reproduce the behavior:
```
git clone .../emsec/hal
mkdir -p hal/{build,pkgdir}
cd hal/build
cmake -DCMAKE_INSTALL_PREFIX=/usr ..
make
make DESTDIR="../pkgdir" install
```

**Expected behavior**
Install should succeed.

**Additional context**
* In source, directory is `deps/spdlog-1.9.2`.  Appears version string `1.5.0` was not updated in `CMakeLists.txt` and `cmake/halConfig.cmake.in`.
* spdlog 1.10.x in Ubuntu apt repository.  1.11.x upstream is available.  When these are installed, is spdlog from `deps` used or ignored?  Is there a flag to use the system-installed version?

joern274, 2023-05-14:
Additionally to the fix provided by @theguy147 I added a cmake line so that libabc gets installed when compiled from deps. By now `make install` works for me.

Must admit that `make install` gets rarely tested since most users start the HAL binary directly from build directory.

xiota, 2023-05-14:
Seems to work now when building from git.

closed 2023-05-14 by xiota