emsec/hal #528: Issues using dataflow plugin

MLDNCL, 2023-08-18:
When I try to import dataflow it could not be found, I opened the fsm example and wanted to try this tool out on it. I am not sure if this was an option I needed to select at some point or file I needed to edit so that this plug in would be available but it doesn't seem to be in the hal_plugins folder or on the GUI plugin display. 

**When I run the following script:**
from hal_plugins import dataflow
res = dataflow.Dataflow.analyze(netlist)

**I receive the following error:**
ImportError: cannot import name 'dataflow' from 'hal_plugins' (/home/mldobs01/hal/build/lib/hal_plugins/__init__.py)

**screen shot of folder for hal_plugins:**
[image]

**screen shot of GUI plugins:**
[image]

RenWal, 2023-08-22:
The dataflow plugin appears to be missing in your install. How did you compile HAL? Did you [enable the plugin at build time](https://github.com/emsec/hal/wiki/Building-HAL#cmake-options)?

MLDNCL, 2023-08-22:
I believe I enabled it but it has been a while since installing it and I can't remember/wasn't planning on using the dataflow plug in at the time, is there an easy way to enable the plugin post install or do I need to uninstall everything and reinstall?

RenWal, 2023-08-24:
If you did not enable the plugin at build time, it was not compiled and is hence unavailable. I believe you don't necessarily  need to reinstall, but it's easiest and you should not lose any data.

If you want to try the manual route, you will still need to recreate your build config with cmake option `-DPL_DATAFLOW=1`, as described in the link from my last comment. Then you invoke `ninja` or `make`, depending on which build system you're using, to compile. This will produce a `lib/hal_plugins/dataflow.so` library in your build directory. You _might_ also get away with only compiling that one library with `ninja dataflow.so` or `make dataflow.so`.
I'm not sure about how you installed HAL. If you used the default install scripts, your plugins should have gone to `/usr/local/lib/hal_plugins/`. You can copy the `dataflow.so` from your build directory here. HAL will pick up any new plugins on launch.

MLDNCL, 2023-08-24:
I am going to reinstall and try enabling the plugins this time and am using the Ubuntu 22.04 system. For this I would just run the following code correct? 

git clone https://github.com/emsec/hal.git && cd hal to clone the Git repository
./install_dependencies.sh
mkdir build && cd build
cmake -DBUILD_ALL_PLUGINS=1
make to compile HAL