emsec/hal #533: pip install

keesj, 2023-09-15:
Hello,

with more recent python and pip versions it is becoming harder and harder to execute pip install specially as root 

As normal user currently (on a recent ubuntu) you either need to use venv after a apt install of python3-venv)

```
python3 -mvenv halenv ; . ./halenv/bin/activate
```
or configuring to bypass the configuration e.g.

cat ~/.config/pip/pip.conf
```
[global]
break-system-packages = true
```

to accommodate for this I suggest removing the "sudo" part and documenting a local install in ~/opt

https://github.com/emsec/hal/blob/77614ece692a521883b488c6f6fbc566fbad21da/install_dependencies.sh#L82C10-L82C10

keesj, 2023-09-15:
For docker.... something like this

RUN apt-get install ... python3-venv and then set 
```
RUN python3 -m venv /opt/halenv
ENV PATH="/opt/halenv/bin:$PATH"
```

joern274, 2023-11-02:
@keesj Thanks for reporting this issue. I am not sure why 'pip3 install' was put into the dependencies-setup-script since the Python modules listed in requirements.txt are available as ubuntu packages as well. Maybe the ubuntu packages where considered to be outdated at the time the decision was made? However, after I double-checked that the required python modules are on the installation list by now I removed the 'pip3 install' line from the ubuntu and docker setup branch.

closed 2023-11-28 by joern274