emsec/hal #542: GUI API does not select gate

i-pere, 2023-11-27:
Hello,

I am launching a python script from the python editor available on HAL GUI.
I followed the examples from [https://github.com/emsec/hal/wiki/Python-GUI-API](url)  using a valid gate ID from my netlist.

Using gui.getSelectedGateIds(), I can retrieve the gate id of the selected gates in the current graph view.
However if I do gui.selectGate(37) or gui.select([netlist.get_gate_by_id(37)]) nothing happens. The previous selection is not cleared and the requested gate 37 is not selected.
The gate 37 is an example, it also does not work on other gates id.

Are there some options to set up ? 

Thanks in advance,

RenWal, 2023-11-28:
Confirmed.

From a cursory look, there might be an issue with how the `GuiApi` interacts with the `SelectionRelay`: the `mAction` is prepared, but apparently never executed. I'm not too familiar with the GUI API, but it appears that `SelectionRelay::relaySelectionChanged` is never called.

joern274, 2023-11-28:
Sorry for stupid bug. Looks like the API was broken since 2021. Glad you found it :smile:

RenWal, 2023-11-29:
Fix in 8a1a1034c841e5767a4a4c8da37466015c079016 tested, looks good to me

closed 2023-11-29 by joern274