emsec/hal #554: How to use functions in igraph quickly

ghyghy123, 2024-01-26:
Hello, I want to use Graph operators in igraph, as shown below, how can I call it quickly in Python command line?
[image]

RenWal, 2024-01-26:
I don't see what would be stopping you from just using the code from this example. (Assuming that this screenshot is showing [this manual page](https://python.igraph.org/en/stable/analysis.html#graph-operators) – please provide links to such sources) The HAL Python console is a fully-featured Python environment. Is this a question about HAL, or a Python question?

Would you mind providing some more information about the issue? What is your Python version? How did you install igraph? What have you tried already, and what exactly doesn't work? Is there any kind of specific error message?

ghyghy123, 2024-01-26:
Thank you very much. The program I entered is as follows, and the error "AttributeError: module 'igraph' has no attribute 'Graph'", is it not installed? Or how do you install igraph?
[image]

RenWal, 2024-01-26:
Huh … This looks like Python imports _something_, but not the _right_ thing. One thing that comes to my mind: Is there, by any chance, a folder called `igraph` in your working directory? And in general, which folder are you in when you start HAL, and what is the command you type in the command line to launch HAL?

ghyghy123, 2024-01-26:
Hello, as shown below, there is a folder called igraph, I start hal in build folder with the command./bin/hal-g,
[image]

RenWal, 2024-01-26:
I guess Python is trying to import that folder as a module, which is not what we want.  
Try going up one folder, and then launching HAL as `./build/bin/hal -g`. See if that resolves the issue.

ghyghy123, 2024-01-26:
Hello, I tried to start in the hal folder, using the command./build/bin/ hal-g, but it still seems to fail, is there any solution?
[image]
[image]

RenWal, 2024-01-26:
Okay, we've solved one problem but there's another issue. I believe you did not install igraph for Python.

Since you have already found the igraph manual pages, please follow the [installation instructions](https://python.igraph.org/en/stable/install.html) there.

ghyghy123, 2024-01-29:
Thank you very much, the igraph package problem was solved by installing it！！！

closed 2024-01-29 by RenWal