emsec/hal #558: Graph algorithm cannot be used

ghyghy123, 2024-02-04:
[image]
Hello, I'm using the sample provided in the graph algorithm, but it says "AttributeError: 'NoneType' object has no attribute 'get_strongly_connected_components' ", I have successfully installed graph algorithm pluging and igraph, is there a solution? Thank you.

RenWal, 2024-02-25:
I can't reproduce this issue. Usually if you get `None` from `get_plugin_instance`, this means that the plugin you've asked for is not available. Please confirm that `graph_algorithm` is available in the [Plugin Manager](https://github.com/emsec/hal/wiki/Plugin-Management), and that the plugin is loaded.

Av0cat0, 2024-03-25:
> I can't reproduce this issue. Usually if you get `None` from `get_plugin_instance`, this means that the plugin you've asked for is not available. Please confirm that `graph_algorithm` is available [...]

I'm having this issue too, I checked the plugin manager and I couldn't find it there:
[image]
However, I couldn't find how to load the graph_algorithm plugin

RenWal, 2024-03-26:
If the plugin is not visible in the plugin manager, then it does not exist in your installation. This is usually due to build options that have disabled this plugin at compile time. See also [this issue comment](https://github.com/emsec/hal/issues/528#issuecomment-1688568231), [the build options documentation](https://github.com/emsec/hal/wiki/Building-HAL#cmake-options) and [the graph algorithms documentation](https://github.com/emsec/hal/wiki/Graph-Algorithms).

Try rebuilding HAL with the `-DPL_GRAPH_ALGORITHM=1` CMake option and see if the plugin shows up.

closed 2024-04-24 by SJulianS