emsec/hal #559: Dockerfile does not work (sudo missing)

shuyelang, 2024-03-05:
The docker file needs a little tweak as ubuntu does not come with sudo installed.

**To reproduce:**

1. docker build . -t hal
2. > ERROR [ 8/12] RUN ./install_dependencies.sh                                                                                                                                                        2.2s 
------                                                                                                                                                                                                      
 > [ 8/12] RUN ./install_dependencies.sh:                                                                                                                                                                   
2.124 ./install_dependencies.sh: line 73: sudo: command not found

** to fix **
update the docker file to do this:

RUN apt-get update -y && \
    apt-get install -y lsb-release && \
    apt-get install sudo -y

RenWal, 2024-03-05:
Good catch on the problem. I would prefer a different solution though.

By default, the build will run as `root` inside the container (we can debate the design choice, but that's a different issue). Instead of installing `sudo` into the container (which should never be needed after the build), the install script should check if it's running as the `root` user, and in this case it should not call `sudo` at all.

RenWal, 2024-03-05:
This appears to be related to a change in how Docker builds packages. With the legacy builder, I can successfully build the image. However, using the [new Buildx engine](https://docs.docker.com/build/architecture/#buildx) (on Docker version 25.0.3, build 4debf41), the build fails.

The `install_dependencies.sh` script is Docker-aware (the check is in line 7), so it should never call `sudo` in the first place if it recognizes that it is running inside a Docker build context. Installing `sudo` inside the container is not the right solution for this problem.

I assume that Buildx does something differently from older Docker versions, which breaks the Docker check.

closed 2024-04-24 by SJulianS