emsec/hal #561: HAL crashes when

CuriousCodeCrafter, 2024-03-11:
The program crashes when I try to create a module on a medium-sized netlist. In my original script, I run a script that gathers a list of all the gates that should be included in a new module that I want to create. The program runs correctly. Then, I run the following code:

... 

m = netlist.create_module(f"module_{i}", netlist.get_top_module())

for gate_id in Gate_IDs:
    m.assign_gate(netlist.get_gate_by_id(gate_id))

without receiving any errors. However, the program automatically shuts down during the recalculation process to display the new design (with the new module) in the "Graph-Views" tab. It appears that although the code runs successfully, there's an issue with redrawing the design in the "Graph-Views" tab after the new module has been created. Both the HAL GUI and the terminal from which I launch the program crash and close automatically, leaving no diagnostic messages.

When I attempt to reopen the project, in the "Modules" section at the upper-left corner I can see that my module has been created and is listed within the top-module hierarchy. However, when the program is about 8% through drawing the design in the "Graph-Views" tab, the program crashes again.

Do you know what can be happening? 

Thanks in advance.

RenWal, 2024-03-11:
There is definitely *something* broken with the module renderer right now, but I couldn't yet pinpoint exactly what it is.

What I found so far:

- Creating a module with a single gate doesn't trigger the issue.
- Creating a module with many gates will *sometimes* cause a crash when the containing module is in an open view (e.g. you create a module under the top module, and you have the graph view of the contents of the top module open)

It appears that the layouter is bailing out with an assertion failure:

    Fatal: ASSERT: "iy1 == iy0 + 1" in file hal/plugins/gui/src/graph_widget/layouters/graph_layouter.cpp, line 781 (hal/plugins/gui/src/graph_widget/layouters/graph_layouter.cpp:781, )

I have built a test case for this which reliably crashes HAL. I'll clean that up a bit and upload it later. Some more familiar with the intricacies of the graph layouter may want to triage based on that.

Also, I found a possibly related issue:

- Deleting a module from the module tree on the top left doesn't redraw the graph view, so the deleted module stays visible. Clicking that module in the graph will then cause HAL to crash since it attempts to access a non-existent module.

RenWal, 2024-03-11:
Executing the script in the following HAL project reliably crashes HAL on an assertion failure in `GraphLayouter::calculateGateOffsets()`.
[hal_project_toy_cipher.zip](https://github.com/emsec/hal/files/14562585/hal_project_toy_cipher.zip)

Just before the crash, `mCoordY` contains the keys 0,2. Hence, the `iy1 == iy0 + 1` assertion in the iteration over this map fails.
I didn't fully figure out how the `mCoordY` map is filled in the first place, perhaps someone else can chime in.

CuriousCodeCrafter, 2024-03-12:
Thank you very much for the help. I have seen that you have updated the corresponding file in the repository. After reinstalling HAL and importing the project where the error previously occurred, I can confirm that the error has been corrected and the modules are now created correctly.

joern274, 2024-03-12:
@CuriousCodeCrafter Thanks for feed. I'll change the issue status to "fixed".

closed 2024-04-24 by SJulianS