emsec/hal #565: Default Plugins Import Error

MatthewLabrada, 2024-05-08:
When attempting to run plugins such as the dataflow analysis or netlist simulator plugins, I get the following error:

<Execute Python Editor content>
ImportError: cannot import name 'dataflow' from 'hal_plugins' (/home/user/hal/build/lib/hal_plugins/__init__.py)

At:
  <string>(2): <module>

I am currently running HAL through WSL2 if that makes any difference. When building I made sure to include the flag '-DBUILD_ALL_PLUGINS=1' and even attempted to build each plugin manually.

Steps to reproduce the behavior:
1. Go to Help
2. Click on Plugins Manager
3. Enable given plugins
4. Back to netlist
5. Attempt to run plugin

I expect normally for the plugins to run without error and provide some sort of output or save a file in a given directory.

OS: Windows 11; WSL (Ubuntu 22.04)

joern274, 2024-05-21:
Hi @MatthewLabrada , thank for your report. Can you please provide the following additional information :

1. Can you verify that the 'dataflow' - plugin has been build and that the shared library is residing in the directory `$HAL_BASE_PATH/lib/hal_plugins` ?
2. When you enable the plugin (your step 3), are there any warnings or errors in the log ?
3. What exactly are you doing at your step 5? If it is a Python script you are running, can you please show the codelines?
4. Is it possible to open a netlist in GUI, enter Plugin Manager again and execute the GUI-extension for 'dataflow'-plugin ?

joern274, 2025-05-09:
Closing issue due to lack of additional information

closed 2025-05-09 by joern274