emsec/hal #578: HAl GUI Hanging

rajatssr835, 2024-06-29:
I am able to successfully installed HAL sing WSL 2. However when I am launching HAL using
sudo ./bin/hal -g

The Gui is hanging and here is my GUI launching messages:
[sudo] password for rajat835:
[core] [info] executing 'hal_gui' with
[core] [info]   '--gui':
Warning: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root' (:0, )
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/bitorder_propagation.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'bitorder_propagation' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/boolean_influence.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'boolean_influence' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/dataflow.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'dataflow' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/gexf_writer.so'...
[29.06.2024 13:48:22] [netlist_writer] [info] registered netlist writer 'Default GEXF Writer' for file extension '.gexf'.
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'gexf_writer' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/graph_algorithm.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'graph_algorithm' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/gui_extension_demo.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'gui_extension_demo' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/hawkeye.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'hawkeye' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/hgl_parser.so'...
[29.06.2024 13:48:22] [gate_library_parser] [info] registered gate library parser 'Default HGL Parser' for file extension '.hgl'.
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'hgl_parser' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/hgl_writer.so'...
[29.06.2024 13:48:22] [gate_library_writer] [info] registered gate library writer 'Default HGL Writer' for file extension '.hgl'.
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'hgl_writer' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/liberty_parser.so'...
[29.06.2024 13:48:22] [gate_library_parser] [info] registered gate library parser 'Default Liberty Parser' for file extension '.lib'.
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'liberty_parser' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/libtest_plugin.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'libtest_plugin' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/netlist_preprocessing.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'netlist_preprocessing' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/netlist_simulator.so'...
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/netlist_simulator_controller.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'netlist_simulator' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/perf_test.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'perf_test' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/python_shell.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'python_shell' loaded upon user request.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/solve_fsm.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'solve_fsm' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/verilator.so'...
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/verilog_writer.so'...
[29.06.2024 13:48:22] [netlist_writer] [info] registered netlist writer 'Default Verilog Writer' for file extension '.v'.
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'verilator' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/verilog_parser.so'...
[29.06.2024 13:48:22] [netlist_parser] [info] registered netlist parser 'Default Verilog Parser' for file type '.v'
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'verilog_parser' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/vhdl_parser.so'...
[29.06.2024 13:48:22] [netlist_parser] [info] registered netlist parser 'Default VHDL Parser' for file type '.vhd'
[29.06.2024 13:48:22] [netlist_parser] [info] registered netlist parser 'Default VHDL Parser' for file type '.vhdl'
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'vhdl_parser' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/waveform_viewer.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'waveform_viewer' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/xilinx_toolbox.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'xilinx_toolbox' loaded to retrieve information about plugin features.
[29.06.2024 13:48:22] [core] [info] loading plugin '/home/rajat835/hal/build/lib/hal_plugins/z3_utils.so'...
[29.06.2024 13:48:22] [gui] [info] GuiPluginManager: 'z3_utils' loaded to retrieve information about plugin features.

Let me know if I need to set any other variable.

RenWal, 2024-07-05:
Running HAL using `sudo` sounds scary. Do you have any particular reason for doing so?

rajatssr835, 2024-07-08:
Than you for your reply. The issue is resolved now. I am using Xlaunch for WSL2 and it solved the issue.

closed 2024-07-10 by RenWal