emsec/hal #588: Failure to build on Arch Linux

xiota, 2024-09-11:
**Describe the bug**

49e9cb8f4977232667c4c816b16474e8e47d402e fails to build on Arch Linux.  Using GCC 14.2.1.  Last successful build was 46d627b3e73 (4.2.0.r355) on 2024-05-08.

Issues:

* Vendored igraph fails to build.  Succeeds when upgraded to 0.10.13. 
    * But still needs system igraph to be installed?
* System igraph is not properly found or used.
    * Probably need to adjust cmake scripts.
* Plugins fail to build.
    * logic evaluator
    * netlist preprocessing
    * resynthesis
    * ze utils

I didn't save logs, so would have to rebuild multiple times to recreate the errors.  Before doing so, should the extra info be added to this issue or new ones?

The igraph issues may be relatively simple to fix, so I'm considering opening PRs.

**To Reproduce**

Try to build [`aur/hal-git`](https://aur.archlinux.org/packages/hal-git) in clean chroot on Arch Linux.  Package needs to be updated, but sufficient to reproduce errors.

Adjust cmake options to observe different errors.  Disable plugins with:
```
    -DPL_LOGIC_EVALUATOR=OFF
    -DPL_NETLIST_PREPROCESSING=OFF
    -DPL_RESYNTHESIS=OFF
    -DPL_Z3_UTILS=OFF
```

**Expected behavior**

Should build successfully.

tmorst84, 2024-09-14:
The same for Ubuntu 22.04.3. Commit 46d627b is the last functional one as well.

xiota, 2024-09-15:
@tmorst84 Would you mind testing this branch?  https://github.com/xiota/hal-emsec/tree/pr_fix

tmorst84, 2024-09-15:
The branch builds and the result is functional. The only problem is that the plugins like dataflow_analysis are not built

xiota, 2024-09-15:
> The only problem is that the plugins like dataflow_analysis are not built

I enabled some previously disabled plugins because they're needed by an enabled plugin.  Other disabled plugins were not changed.  To build all plugins, add option to cmake when configuring: `-DBUILD_ALL_PLUGINS=ON`

xiota, 2025-05-19:
Closing because specific issues identified are no longer relevant.

closed 2025-05-19 by xiota