emsec/hal #590: No "netlist" error when running a Python script on a netlist in commandline mode

harnhua, 2024-09-23:
Hi,
After successfully loading a netlist and executing the following lines of sample Python code in the Python widget, I saved the commands in a .py script.

```
from hal_plugins import graph_algorithm

graph = graph_algorithm.NetlistGraph.from_netlist(netlist)
```

However, when I tried to load the project and execute the same commands in a .py script, HAL says that the "netlist" object does not exist.

`hal --project-dir <directory containing HAL project> --python-script test.py
`
Looking at the log messages, it seems like the Python script was executed before the netlist was loaded.

Am I missing an argument or Python command?

Thanks!

harnhua, 2024-09-24:
If I omit the --python-script argument, then the project seems to load without issue.
`hal --project-dir <directory containing HAL project>`

Is there a way to make the Python script execute after the project is loaded?
Or alternatively, is there a command to load the project from the Python script?

RenWal, 2024-10-14:
The old `python_shell` plugin wasn't updated in about 5 years and is therefore not really compatible with the current state of HAL. You do still get a Python shell, but don't have access to any of the project context, so it's not very useful.

The file that needs to be updated is https://github.com/emsec/hal/blob/c6f7150f5130549ffe62b8a95cc2decfa3af0c78/plugins/python_shell/src/plugin_python_shell.cpp . Pull requests are welcome. You can look at the way that the GUI initializes its Python interpreter, particularly the `PythonContext::initializeScript` invocation:
https://github.com/emsec/hal/blob/ed28db50071c93f61d301880ecd735ae17302dd4/plugins/gui/src/python/python_context.cpp#L154