emsec/hal #606: RHEL Build

Blebowski, 2025-03-07:
Hi,

would it be possible to support RHEL build please ?

Lot of commercial EDA tools work on RHEL, therefore companies use it (or CentOS, Rocky).

I was trying to give it a shot on RHEL8, and figured out couple of issues:
- z3 is not packaged for RHEL -> Must be built from source and installed
- Older GCC versions have trouble on some C++17 features -> Enforcing CLANG 17 helped
- Some syntax errors occur since `std::setw` is not visible in couple of files -> Needs include of `<bitmanip.h>`.
- Can't find Python11 (currently our machine has 3.6 as default, and a main virtual env has 3.8)

Actionable points from this would be:
- Extend `install_dependencies` script to install the RHEL package equivalents
- Clone and build Z3 on RHEL as it is not packaged.
- Extend build scripts to provide path to python libraries (so that attempt to link to 3.11 can be overridden by other Python version).

joern274, 2025-05-12:
Experimental support for RHEL added by now. Tested on RHEL 9.5 only, not sure whether it also works on RHEL 8. Awaiting test results from RHEL user. Will close this issue unless receiving feedback how to improve RHEL support.

Blebowski, 2025-05-12:
I will try this on the RHEL machine and let you know.

joern274, 2025-07-15:
RHEL refused to extend the trial period to give us the chance to help RHEL users in the future. I hope the provided solution gets you up running - if not, there is very little I can do. 🥲

closed 2025-07-15 by joern274