emsec/hal #612: Cloning HAL takes long

Blebowski, 2025-05-12:
**Describe the bug**
When I clone this repo, it is alltogether a 3.6 GB repo. It takes a while to clone, especially on WIFI.
Out of it, 3.3 GB is the `.git` folder. What is the reason for it ? Could there were large binary files commited
long ago ? Possibly these are not needed anymore and could be removed...

joern274, 2025-05-13:
Short story: please use `git clone --depth 1 https://github.org/emsec/hal` to omit the history.

Long story: the size of the `.git` folder is the result of ten years history of contributions times the individual file size which is quite large for some gate libraries, sample netlists and screenshots for documentation. We will never temper with this history since we need the information from time to time, e.g. to reconstruct the state of HAL when a paper was submitted or to list all contributions of a team member. This said it relatively easy to omit that information if not needed and I use the command above myself when downloading HAL to a VM with limited disk size.

Blebowski, 2025-05-21:
OK, thanks for explanation.

closed 2025-05-21 by Blebowski